boltgolt/howdy #225: Fedora 30 config

msandy1, 2019-06-29:
I can't seem to get it to run on Fedora 30. i've been able to set the device path. /dev/v4l/by-path/[DEVICE]. but it's still not running for me. everytime i test it is get and error saying :

[ INFO:0] VIDEOIO: Enabled backends(7, sorted by priority): GSTREAMER(1000); V4L2(990); OPENNI(980); CV_IMAGES(970); CV_MJPEG(960); FIREWIRE(950); UNICAP(940)
OpenCV | GStreamer warning: GStreamer: unable to start pipeline
 (/builddir/build/BUILD/opencv-3.4.4/modules/videoio/src/cap_gstreamer.cpp:872)

(python3:18466): GStreamer-CRITICAL **: 15:46:21.361: gst_element_get_state: assertion 'GST_IS_ELEMENT (element)' failed
OpenCV | GStreamer warning: GStreamer: unable to query pipeline state (/builddir/build/BUILD/opencv-3.4.4/modules/videoio/src/cap_gstreamer.cpp:420)
libv4l2: error setting pixformat: Input/output error
VIDEOIO ERROR: libv4l unable to ioctl S_FMT
VIDEOIO ERROR: libv4l unable to ioctl VIDIOCSPICT

CvCapture_OpenNI::CvCapture_OpenNI : Failed to open input file (/dev/v4l/by-path/pci-0000:00:1d.7-usb-0:5:1.2-video-index0): Bad file extension

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

No protocol specified
qt.qpa.xcb: could not connect to display :0
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, xcb.

Aborted

I've tried to reinstall xcb but that made no difference? is howdy working with fedora 30. is there a default recommended post install configuration?

hamomel, 2019-07-09:
I managed to make it run on Fedora 30, but it only works with sudo not on login screen. I found that ir-camera on my Thinkpad T580 is on /dev/video0 and set device_path = /dev/video0. Also I use Fedora wits Xorg not Wayland. I have the same warnings from GStreamer, but after these warnings I get "Identified face my_user_name" and log in as su.
Does anyone know how to make it start on login screen?

boltgolt, 2019-09-16:
Sorry, your issue was earlier but to keep the discussion in 1 place i'll mark this as a duplicate of #233

closed 2019-09-16 by boltgolt