boltgolt/howdy #230: sudo howdy test fails on Wayland

ZeroCool2u, 2019-07-08:
_Please describe the issue in as much detail as possible, including any errors and traces_

On a Dell XPS 13 running Ubuntu 19.04 with Wayland  running `sudo howdy test` generates the following error: 

`Invalid MIT-MAGIC-COOKIE-1 key: cannot connect to X server :0`

I suspect that this is an issue with OpenCV attempting to capture frames with a method that isn't supported without an x-server available, because I'm running Ubuntu on Wayland. I've tried running it with `QT_QPA_PLATFORM=wayland` as well with the same result. 

It also may be related to #104 

Note, howdy works perfectly otherwise and detected the correct (infrared) webcam with zero configuration. :smile: 

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu 19.04

Howdy version: 2.5.1

picaschaf, 2019-07-26:
Same problem here on Fedora 30.

simonbcn, 2019-07-28:
This app doesn't work with Wayland.

ZeroCool2u, 2019-07-28:
If you look at the issue above you'll see it definitely does work with Wayland. The test command fails on Wayland though.

boltgolt, 2019-08-02:
It is related to #104, with the same cause, sudo fails to find the X server. Take a look at this [reddit thread](https://old.reddit.com/r/Fedora/comments/5eb633/solution_running_graphical_app_with_sudo_in/).

berndso, 2019-08-20:
It seems that this will be fixed in GNOME 3.34 https://www.phoronix.com/scan.php?page=news_item&px=XWayland-Xauth-GNOME-Works

ardevd, 2020-05-15:
So what's the conclusion? Will it work fine under Wayland apart from `howdy test`? I'm running Sway so fixes for GNOME does not help me.

ZeroCool2u, 2020-05-26:
@ardevd You'll probably just need to try it yourself since you're using Sway and not GNOME. Setup is pretty painless though, shouldn't be more than a couple minutes to setup.

ardevd, 2020-05-26:
I can confirm it works just fine!

boltgolt, 2020-09-28:
Closing this as a stale issue

closed 2020-09-28 by boltgolt

Zocker1999NET, 2020-10-02:
I'm running sway (installed from Debian Unstable Repo) and newest version from howdy (2.6.1) but still got this issue:
```
$ sudo howdy test
Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

No protocol specified
Unable to init server: Could not connect: Connection refused

(python3:1199520): Gtk-WARNING **: 18:46:56.392: cannot open display: :0
```

Got no problems with other apps, Firefox is using Wayland native, but Xwayland is also enabled and DISPLAY var is set correctly as other apps like `xeyes` can be opened. It would be fine if `howdy test` would even open using Xwayland, don't need native Wayland support on this one.

@ardevd Have you configured anything special to let `howdy test` run?

Zocker1999NET, 2020-10-02:
Sorry, found my issue after trying `sudo xeyes`: Sway's Xwayland seems to block access for root applications by default, fixed with `xhost local:root`, see https://askubuntu.com/questions/175611/cannot-connect-to-x-server-when-running-app-with-sudo