boltgolt/howdy #231: Debian Samsung Notebook 9 Pro

betterthanyou710, 2019-07-09:
Hey,
I'm right now not a big programmer and I'm running Debian on my Samsung but Howdy keeps spitting out an error I don't really understand and therefore don't know how to deal with. All of my troubleshooting ideas have failed so far and I'll keep trying but I wanted to see if anyone else in the community had the same problem. The problem arises when I try to add a face model. This computer has 2 cameras and 2 IR scanners and I think they pathed automatically but I'm not sure.

Here is the error message:
`Enter a label for this new model [Initial model] (max 24 characters): Ben
select timeout
VIDIOC_DQBUF: Resource temporarily unavailable

Please look straight into the camera
VIDIOC_QBUF: Invalid argument
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/lib/security/howdy/cli/add.py", line 147, in <module>
    gsframe = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.1.0) /io/opencv/modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'
`
If there is any other information that would be helpful, please let me know.

Thanks so much for the help!

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Debian

Howdy version: 2.5.1

edsadr, 2019-07-26:
I have the same issue in a Razer Blade Stealth 13" 

```
Adding face model for the user edsadr
Enter a label for this new model [Initial model] (max 24 characters): 
select timeout
VIDIOC_DQBUF: Resource temporarily unavailable

Please look straight into the camera
VIDIOC_QBUF: Invalid argument
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 147, in <module>
    gsframe = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.1.0) /io/opencv/modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'
```

boltgolt, 2019-08-02:
"Resource temporarily unavailable" could be a USB error, does the camera work in other programs (such as Cheese)?

edsadr, 2019-08-05:
yes it works in Cheese and other apps like Zoom, Slack and Google Hangouts

edsadr, 2019-08-05:
also, when trying to create the mode you can see the camera and the infrared working forI would say 10 seconds then it shows the errors described at the top, any way to debug this to provide more insight @boltgolt ?

edsadr, 2019-08-12:
ok fixed this, was a config issue I have these devices:

```
ls /dev/v4l/by-path/
pci-0000:00:14.0-usb-0:6:1.0-video-index0  pci-0000:00:14.0-usb-0:6:1.0-video-index1  pci-0000:00:14.0-usb-0:6:1.2-video-index0  pci-0000:00:14.0-usb-0:6:1.2-video-index1
``` 

Config was pointing to `pci-0000:00:14.0-usb-0:6:1.2-video-index0` and I changed to `pci-0000:00:14.0-usb-0:6:1.0-video-index0`, then ran the test and now it works like a charm

edsadr, 2019-08-12:
the thing is that the infrared doesn't turn on, so it requires light to work when it starts turning a bit dark then it stops identifying my face

karanjot21, 2019-08-29:
VIDEOIO ERROR: V4L2: Pixel format of incoming image is unsupported by OpenCV

Please look straight into the camera
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/lib/security/howdy/cli/add.py", line 147, in <module>
    gsframe = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.1.0) /io/opencv/modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'

Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 145, in apport_excepthook
    os.O_WRONLY | os.O_CREAT | os.O_EXCL, 0o640), 'wb') as f:
FileNotFoundError: [Errno 2] No such file or directory: '/var/crash/_lib_security_howdy_cli.py.0.crash'

Original exception was:
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/lib/security/howdy/cli/add.py", line 147, in <module>
    gsframe = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.1.0) /io/opencv/modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'

boltgolt, 2019-11-11:
@edsadr there are some developments for laptops with IR blasters that don't turn on, check out #269 

@karanjot21 that's a completely different (and fixed) issue, see #88

closed 2021-04-02 by boltgolt