boltgolt/howdy #232: Grey screen while testing

PancakeSparkle, 2019-07-14:
So, I decided to try howdy again, this time on Arch, since I heard 2.5 was supposed to be compatible with more cameras, this time doing howdy test results in a grey screen, force_mjpeg is on, I tried playing around with some settings and nothing worked, my camera is an SR300, aka the Razer Stargazer. I am on the latest kernel, which is currently 5.2. 
[image] This is what it looks like when I do a test, the IR led seems to turn on, I have tried all the video devices on my system. When I do a `v4l2-ctl -d /dev/video3 --list-formats` this comes up: `ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'YUYV' (YUYV 4:2:2)
	[1]: 'Z16 ' (16-bit Depth)
	[2]: 'INZI' (Planar 10:16 Greyscale Depth)
	[3]: '' (435a4e49-b602-480f-978c-e4e88ae)
	[4]: 'Y10 ' (10-bit Greyscale)
	[5]: '' (52494150-8536-4841-b6bf-8fc6ffb)`

boltgolt, 2019-08-02:
Does the camera work in any other programs? Changing resolutions could also help

PancakeSparkle, 2019-08-02:
RGB does work in other programs, IR doesn't, well, except in the realsense viewer, other programs show a green screen, realsense-viewer's the only one that can show the actual IR feed, the resolution it uses is 640x480 by default, with Y8 and Y16 available as formats

boltgolt, 2019-08-02:
Unfortunately that probably means that Howdy will not work with your camera, because Howdy depends on linux driver support. You could try switching the recorder in the config, but i don't think it will fix it

You can always use Howdy with the normal camera, but it will not be as secure

PancakeSparkle, 2019-08-02:
somehow realsense-viewer works with it, there's patches for uvcvideo by intel, but they're for kernel 4.x, not 5

PancakeSparkle, 2019-08-02:
when the grey screen comes up, the red dot on the IR emitter turns on, so it's trying

cydia2020, 2020-03-26:
> RGB does work in other programs, IR doesn't, well, except in the realsense viewer, other programs show a green screen, realsense-viewer's the only one that can show the actual IR feed, the resolution it uses is 640x480 by default, with Y8 and Y16 available as formats

IR Feed can be correctly displayed in Google Duo, if you wanna test this out, go to https://duo.google.com/ and select the IR camera as your input.

closed 2021-04-02 by boltgolt