boltgolt/howdy #233: howdy does not work in Fedora 30

shatadru, 2019-07-14:
_Please describe the issue in as much detail as possible, including any errors and traces_

The installation works fine. After I set device_path=/dev/video0 it allows me to do "howdy add"
however, it is not activated when I try to login.

Linux distribution (if applicable):
Fedora Workstation 30

Howdy version:
 rpm -q howdy
howdy-2.5.1-2.fc30.x86_64

Fedora 30 Cinnamon desktop. 

I suspect the issue is related to pan config.

iamzubin, 2019-07-14:
can confirm this issue.

dill-wood, 2019-07-30:
Can confirm this is an issue for me as well. Program seems to install correctly. No errors displayed. Can successfully run 'sudo howdy test', but program does not seem to be registering active for the login screen and the terminal prompt.

iamzubin, 2019-08-14:
did anyone get a workaround?

dill-wood, 2019-08-15:
Not so far. Was seeing if @boltgolt might have an idea when he had the chance. Wasnt in a huge rush :)

Hakun1n, 2019-08-19:
Same for Fedora29 (i did not upgraded yet). Test works but no camera/IR gets activated (no red squeres in IR emmiters) during sudo nor logon. 

Thank you in advance for fixing this!

Alex

alfarock, 2019-09-09:
Can confirm. 

Fedora 30
Dell XPS 9370 

"sudo howdy test" works on X11 but I get the following:

[ INFO:0] VIDEOIO: Enabled backends(7, sorted by priority): GSTREAMER(1000); V4L2(990); OPENNI(980); CV_IMAGES(970); CV_MJPEG(960); FIREWIRE(950); UNICAP(940)
OpenCV | GStreamer warning: GStreamer: unable to start pipeline
 (/builddir/build/BUILD/opencv-3.4.4/modules/videoio/src/cap_gstreamer.cpp:872)

(python3:7282): GStreamer-CRITICAL **: 21:30:52.997: gst_element_get_state: assertion 'GST_IS_ELEMENT (element)' failed
OpenCV | GStreamer warning: GStreamer: unable to query pipeline state (/builddir/build/BUILD/opencv-3.4.4/modules/videoio/src/cap_gstreamer.cpp:420)

Had to add "/dev/video2" to the config.ini file to get "howdy test" working.  I also tried adding "auth sufficient pam_python.so /lib64/security/howdy/pam.py" to the files in /etc/pam.d  but that didn't work.

PsiACE, 2019-09-16:
Can confirm.

- Fedora 30 
- Gnome 3.32
- HP Elitebook 735G5

Additional information.

```
No protocol specified
qt.qpa.xcb: could not connect to display :0
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, xcb.
```

Does KDE work properly?

itstuffed, 2019-09-26:
Same here
Output of 

`sudo howdy test`

```
[ INFO:0] VIDEOIO: Enabled backends(7, sorted by priority): GSTREAMER(1000); V4L2(990); OPENNI(980); CV_IMAGES(970); CV_MJPEG(960); FIREWIRE(950); UNICAP(940)
OpenCV | GStreamer warning: GStreamer: unable to start pipeline
 (/builddir/build/BUILD/opencv-3.4.4/modules/videoio/src/cap_gstreamer.cpp:872)

(python3:4469): GStreamer-CRITICAL **: 22:28:12.366: gst_element_get_state: assertion 'GST_IS_ELEMENT (element)' failed
OpenCV | GStreamer warning: GStreamer: unable to query pipeline state (/builddir/build/BUILD/opencv-3.4.4/modules/videoio/src/cap_gstreamer.cpp:420)

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

No protocol specified
qt.qpa.xcb: could not connect to display :0
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, xcb.

Aborted
```

dill-wood, 2019-10-07:
> This part
> 
> > No protocol specified
> > qt.qpa.xcb: could not connect to display :0
> > qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
> > This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
> 
> > Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, xcb.
> 
> Sounds like you're running Wayland when Howdy (or one of its dependencies) expects you to be using X11. When you log in (for the first time after boot) click on the gear next to the login button and select the option that contains `Xorg` or `X11`. Do you still get the same error?
> 
> Fedora 30 (I think) started defaulting to running GNOME on Wayland instead of X11, so it would make sense for this to be the reason for the breakage.

Cant speak for him, but I know that all testing and attempts to get Howdy to work in Fedora 30 for me was all on X11 and I still ran into issues.

Doridian, 2019-10-11:
I got it to work for `sudo -i` by manually adding it to the `/etc/pam.d/system-auth` file and installing pam_python.so (which can be found in the RPM Sphere repository).
This situation is not ideal at all as pam-python still requires python 2, which basically no distro likes to maintain at this point (due to it going EoL very soon)

Here is my pam.d file that I referred to (so you see where it must be inserted)
```
doridian@fennec ~]$ cat /etc/pam.d/system-auth 
#pam.d Generated by authselect on Thu Oct 10 17:51:17 2019
# Do not modify this file manually.

auth        required                                     pam_env.so
auth        required                                     pam_faildelay.so delay=2000000
auth        sufficient                                   pam_fprintd.so
auth        sufficient                                   pam_python.so /lib64/security/howdy/pam.py
auth        [default=1 ignore=ignore success=ok]         pam_succeed_if.so uid >= 1000 quiet
....
```

vladimirtravalja, 2019-10-11:
Its working under Wayland, Fedora 31 beta, Gnome 3.34 and PAM set as Doridian said (only terminal root access).
The only thing I had to find which source camera is so in my case is /dev/video0 and I had to install older rpm from fedora30 pam_python as 31 is still in testing phase.
but nevertheless everything works fine.

dill-wood, 2019-10-11:
So in theory,

If us gents who are on 30 wait until the official 31 release and then install from the appropriate repo, the issue should be resolved?

Thanks,

vladimirtravalja, 2019-10-11:
As I mentioned above, it works on f31, Wayland, kernel 5.4-rc2, with pam_python binary for fc30, only if you set it correctly.

vladimirtravalja, 2019-10-26:
One problem found. When you try to open an application such as "timeshift" which initially requires a password to be executed, video login does not work, in fact it looks like howdy calls camera to take a photo, but ends in a loop, as my camera constantly flickers and trying to get picture (possibly some issue with PAM, but again no output so I cannot tell). Not sure what is the cause of that, but apparently GUI password inquiry and Howdy does not work properly. Again, opening application such as Timeshift executes password inquiry and it gives directive to howdy to execute camera function, but ends in loop and application "timeshift"  is never starts...

DnsZhou, 2019-11-21:
Not working on Fedora 31. Similarly, `sudo howdy test` works with the following output:
```
[ INFO:0] VIDEOIO: Enabled backends(7, sorted by priority): GSTREAMER(1000); V4L2(990); OPENNI(980); CV_IMAGES(970); CV_MJPEG(960); FIREWIRE(950); UNICAP(940)
OpenCV | GStreamer warning: GStreamer: unable to start pipeline
 (/builddir/build/BUILD/opencv-3.4.6/modules/videoio/src/cap_gstreamer.cpp:872)

(python3:12491): GStreamer-CRITICAL **: 14:29:47.448: gst_element_get_state: assertion 'GST_IS_ELEMENT (element)' failed
OpenCV | GStreamer warning: GStreamer: unable to query pipeline state (/builddir/build/BUILD/opencv-3.4.6/modules/videoio/src/cap_gstreamer.cpp:420)

Opening a window with a test feed
```

Fedora 31
Gnome 3.34.1
Lenovo Thinkpad T590

DnsZhou, 2019-11-21:
It works with me on Fedora 31 & Wayland with solution from @Doridian , Thanks a lot.

- Fedora 31
- GNOME 3.34.1
- Lenovo Thinkpad T590
### Steps: 

- Before that:
-- Install **howdy** according to the instruction.
-- Run `sudo howdy config` and set device_path to the camera path of your device, e.g., `device_path = /dev/video0` (You may need to install **nano editor** if you got FileNotFoundError, see #290 )

1.  Install **RPM Sphere repository**
  Follow  https://rpmsphere.github.io/

2.  Install **pam_python** (not python_pam) by the following command:
`sudo yum install pam_python`

3.  Manually edit **/etc/pam.d/system-auth** file and make it like this:

```
user@localhost ~]$ sudo vi /etc/pam.d/system-auth 

#pam.d Generated by authselect on Thu Oct 10 17:51:17 2019
# Do not modify this file manually.

auth        required                                     pam_env.so
auth        required                                     pam_faildelay.so delay=2000000
auth        sufficient                                   pam_fprintd.so
auth        sufficient                                   pam_python.so /lib64/security/howdy/pam.py
auth        [default=1 ignore=ignore success=ok]         pam_succeed_if.so uid >= 1000 quiet
....
```
Where **auth        sufficient                                   pam_python.so /lib64/security/howdy/pam.py** is a newly inserted line to the file.

4. Run `sudo howdy test ` and `sudo -i` to check whether it works.

in1t3r, 2019-11-30:
for the howdy to work you need to edit some of your pam.d configuration files if they are not edited by the installer. 
Basically you will want to edit the `/etc/pam.d/sudo` and add the following line:
`auth sufficient pam_python.so /lib/security/howdy/pam.py`
at the beginning so that authentication system first try a facial recognition and if that works it will grant you privilege if not it will start other methods of authenticating namely password in most cases.

You can follow the archwiki https://wiki.archlinux.org/index.php/Howdy on how to setup.
Also if you are using for example GDM or LightDM you would want to edit their configuration files inside  `/etc/pam.d/` and add the same line. 
Yeah its manual setup but at least you control where you want to add and where you don't want to add.

bkmgit, 2020-01-14:
For Fedora, please consider testing the repo at:
https://copr.fedorainfracloud.org/coprs/luya/howdy/

EfraimLA, 2020-01-27:
> The sudo authentication works with me on Fedora 31 & Wayland with solution from @Doridian , Thanks a lot.
> 
> * Fedora 31
> * GNOME 3.34.1
> * Lenovo Thinkpad T590
> 
> ### Steps:
> * Before that:
>   -- Install **howdy** according to the instruction.
>   -- Run `sudo howdy config` and set device_path to the camera path of your device, e.g., `device_path = /dev/video0` (You may need to install **nano editor** if you got FileNotFoundError, see #290 )
> 
> 1. Install **RPM Sphere repository**
>    Follow  https://rpmsphere.github.io/
> 2. Install **pam_python** (not python_pam) by the following command:
>    `sudo yum install pam_python`
> 3. Manually edit **/etc/pam.d/system-auth** file and make it like this:
> 
> ```
> user@localhost ~]$ sudo vi /etc/pam.d/system-auth 
> 
> #pam.d Generated by authselect on Thu Oct 10 17:51:17 2019
> # Do not modify this file manually.
> 
> auth        required                                     pam_env.so
> auth        required                                     pam_faildelay.so delay=2000000
> auth        sufficient                                   pam_fprintd.so
> auth        sufficient                                   pam_python.so /lib64/security/howdy/pam.py
> auth        [default=1 ignore=ignore success=ok]         pam_succeed_if.so uid >= 1000 quiet
> ....
> ```
> 
> Where **auth sufficient pam_python.so /lib64/security/howdy/pam.py** is a newly inserted line to the file.
> 
> 1. Run `sudo howdy test ` and `sudo -i` to check whether it works.
> 
> With these steps above, the sudo authentication works perfectly. But login page authentication still not working. I tried to add similar configuration to /etc/pam.d/password-auth, but it became even worse: I can not even log in with a right password, similar with #208
> 
> Another interesting thing I found was when using the virtual terminal (ctrl + alt + F2) and trying to log in, howdy popped out following error messages:
> 
> ```
> VIDEOIO ERROR: V4L: device /dev/video0: Unable to query number of channels
> CvCapture_OpenNI::CvCapture_OpenNI : Failed to initialize the context: The file is corrupted!
> Could not read from camera  
> ```
> 
> But it works after I logged in manually. I tried `chmod o+x /lib64/security/howdy/` and `chmod -R 755 /lib64/security/howd`, neither of them works.
> 
> Is there any solution on that?

That work for me but i cannot use it for login my account, do i have to edit the PAM Config again for that?

thoroc, 2020-04-12:
Followed all the steps (add `pam_python`, `chmod +x` howdy dir, set the `device_path`, install `nano`) and still getting the same error on test.

Fedora 31

```
[ INFO:0] VIDEOIO: Enabled backends(6, sorted by priority): GSTREAMER(1000); V4L2(990); OPENNI(980); CV_IMAGES(970); CV_MJPEG(960); FIREWIRE(950)
[ WARN:0] OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] OpenCV | GStreamer warning: GStreamer: pipeline have not been created

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
```

principis, 2020-05-03:
@thoroc that error is related to #318 you can ignore it if everything works.

@EfraimLA Are you using `sddm` for logging in? 
The issue for me was related to selinux.
my config:
```
$ cat /etc/pam.d/sddm
auth     [success=done ignore=ignore default=bad] pam_selinux_permit.so
auth       sufficient   pam_python.so /lib64/security/howdy/pam.py
...
```
To fix the issue:
```
ausearch -c 'python3' --raw | audit2allow -M my-python3
semodule -i my-python3.pp
```

lightonflux, 2020-06-03:
Fedora 30 is end of life. 
https://lists.fedoraproject.org/archives/list/devel@lists.fedoraproject.org/message/7UTUFY7WEL6RTFRXJB75XAFH44Y6RPUC/

Is this issue still relevant to newer versions?

thoroc, 2020-06-23:
I am still on Fedora 31 (32 is already out) and the issue persist

baelter, 2021-06-10:
Have same (or similar) issue on fedora 34. `howdy add` and `howdy test` works fine. Updated the `etc/pam.d/sudo` file but I just get password prompt when I do `sudo -i`

KoalaV2, 2021-06-21:
> Same here
> Output of
>
> `sudo howdy test`
>
> ```
> [ INFO:0] VIDEOIO: Enabled backends(7, sorted by priority): GSTREAMER(1000); V4L2(990); OPENNI(980); CV_IMAGES(970); CV_MJPEG(960); FIREWIRE(950); [...]

I'm having the same issue here. Howdy works perfectly fine on Arch Linux with Sway installed other that running `sudo howdy test` crashes due to the errors written in this comment.

principis, 2021-06-22:
@KoalaV2 you should open a new issue with more information, because that has nothing to do with the OP's issue.

This issue should actually be closed.

gzqx, 2021-08-24:
I am using Fedora Plasma 34 (which is the KDE version of fedora). After doing what @Doridian prompt to make sudo work, I also managed to make login authentication work by adding the same line into `password-auth` below the first two "required" line.

closed 2024-04-09 by musikid