boltgolt/howdy #235: Unable to install on kubuntu 19.04 (same issue on kde neon)

thondal, 2019-07-23:
_Please describe the issue in as much detail as possible, including any errors and traces_
Trying to install howdy following Readme, but recieve this error:

```
Traceback (most recent call last):
  File "/var/lib/dpkg/tmp.ci/preinst", line 54, in <module>
    devices = os.listdir("/dev/v4l/by-path")
FileNotFoundError: [Errno 2] No such file or directory: '/dev/v4l/by-path'
dpkg: error processing archive /var/cache/apt/archives/howdy_2.5.1_all.deb (--unpack):
 new howdy package pre-installation script subprocess returned error exit status 1
An error occured during the processing of:
 /var/cache/apt/archives/howdy_2.5.1_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):
Kubuntu 19.04
KDE Neon (latest)

Howdy version:
2.5.1

boltgolt, 2019-08-02:
Do you have V4L2 installed? You don't seem to have the `/dev/v4l/by-path` folder, which i have never seen before

thondal, 2019-08-04:
Seems to be the same issue as: https://github.com/boltgolt/howdy/issues/137
I also have a surface book, so probably won't work then.

kiraitachi, 2020-01-01:
I just got the same problem.....It was working a couple days for me and now it stopped working. After trying to reinstall I get this error...Im on Pop OS latest version 19.10.

closed 2021-04-02 by boltgolt