boltgolt/howdy #237: Error installing Howdy: can't find webupd8team dependency

jacdavwill, 2019-07-28:
I get the following error when running the command ```sudo add-apt-repository ppa:boltgolt/howdy```

Err:14 http://ppa.launchpad.net/webupd8team/gnome3/ubuntu disco Release
  404  Not Found [IP: 91.189.95.83 80]
Reading package lists... Done
E: The repository 'http://ppa.launchpad.net/webupd8team/gnome3/ubuntu disco Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

Linux distribution (if applicable): Pop!_OS 19.02 (this may be the root of the problem)

Howdy version: the current one I assume

wengj9, 2019-07-30:
I believe the issue isn't related to `howdy`, it's another repository on your system. The `webupd8team/gnome3` repo hasn't been updated in several years, not sure why you have it added. The quick and dirty way to fix the error should be `sudo apt install ppa-purge` and `sudo ppa-purge ppa:webupd8team/gnome3`.

boltgolt, 2019-08-02:
Thanks for the help @wengj9, good to read that this was fixed so quickly

closed 2019-08-02 by boltgolt