boltgolt/howdy #238: Failing to authenticate on lockscreen

NICHOLAS85, 2019-08-02:
_Please describe the issue in as much detail as possible, including any errors and traces_

Setup: As the "streamer" package is not available in Debian testing I created a faux package using equivs which fufills this dependency and then installed Howdy using the following command:

		HOWDY_NO_PROMPT=1 sudo -E dpkg -i howdy_2.5.1.deb

I then set the device_path variable and added a face model.

The installation was successful and Howdy works correctly for sudo and executing programs which require elevated rights.

The problem occurs when attempting to unlock my device. I see my infrared cameras turn on for a moment, much shorter than when attempting to use sudo, and then the computer fails to unlock.

Running

		/usr/lib/x86_64-linux-gnu/libexec/kscreenlocker_greet --testing

Shows the following console messages:

		Data files have not been downloaded, please run the following commands:
		
		        cd /lib/security/howdy/dlib-data
		        sudo ./install.sh
		
		Traceback (most recent call last):
		  File "/lib/security/howdy/compare.py", line 219, in <module>
		    face_locations = face_detector(gsframe, 1)
		TypeError: 'NoneType' object is not callable
		Authentication failure

Running

		tail /var/log/auth.log

Shows

		kcheckpass[4363]: pam_unix(kde:auth): Couldn't open /etc/securetty: No such file or directory
		kcheckpass[4344]: pam_unix(kde:auth): authentication failure; logname=nicholas uid=1000 euid=1000 tty=/dev/pts/1 ruser= rhost=  user=nicholas

With the first line being repeated 2-6 times before authentication failure message.

I've run sudo ./install.sh in /lib/security/howdy/dlib-data and it shows that the data files have been downloaded.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Netrunner (Debian Testing based distribution)

Howdy version: 2.5.1

boltgolt, 2019-08-02:
The `/etc/securetty` issue should be fixed by touching an empty file there, but i don't think that's what's breaking Howdy.

The datafiles being there (while Howdy is warning that they aren't) might point to Howdy not having the right permissions to read them. If not, it could be that the PATH is wrong. You could try to print the PATH by adding a print statement at [line 61 in compare.py](https://github.com/boltgolt/howdy/blob/master/src/compare.py#L61).

NICHOLAS85, 2019-08-02:
Printing the path at line 61 returns "/lib/security/howdy".
What permissions should be present on the dlib-data folder? Currently, it's set with drwxr--r--

NICHOLAS85, 2019-08-02:
It did seem to be a permission issue. Running

`chmod o+x dlib-data/`

fixed the issue.

closed 2019-08-02 by boltgolt

noonedeadpunk, 2020-04-27:
> It did seem to be a permission issue. Running
> 
> `chmod o+x dlib-data/`
> 
> fixed the issue.

Thanks, resolved the issue for my sddm lock screen as well

Chimchar10, 2020-05-11:
I changed the permissions for the dlib-data folder but even then it seems like my camera just doesn't recognize that there's a face at all at the lock screen. When I go into sudo -i it does it very quick.

nekromoff, 2020-12-03:
came here from neon 20.04, same issue

siddhartpai, 2020-12-27:
From [https://github.com/boltgolt/howdy/issues/450](#450) setting 

sudo chmod a+x /lib/security/howdy/recorders/  solves the issue at the lock screen.

iiegn, 2022-01-25:
For my screensaver (mate-screensaver) under Ubuntu 20.04, with Howdy version 2.6.1 to work, I needed *both*
`sudo chmod o+x /lib/security/howdy/{dlib-data,recorders}`

(this is a summary of the separately mentioned hints from above - it took me a moment to realise that there are two different directories mentioned - and I needed both.)