boltgolt/howdy #239: howdy add hangs and no IR LEDs on Lenovo X395

j-m-p, 2019-08-02:
On the Lenovo Thinkpad X395 I seem to have a combination camera module with IR and Color modes.
v4l presents 4 devices. When I use cheese, all 4 of them present a color image. But the debian preinst script identifies two of them as "Integrated Camera: Integrated I" and two as "Integrated Camera: Integrated C".

The image of the "I" types from howdy test looks like a b/w with a bit of a halo around the glasses so I think this could actually be IR but the LEDs don't come on. Resolution on this one is 360x640.
When I pick the "C" type camera I get a normal b/w image with 480x640.

howdy add doesn't recognize a face on the IR cam but does on the Color cam. Presumably also because it's too dark without the IR LED.

The IR image is quite dark and I actually had to set the dark_threshold to 100 so howdy add won't hang indefinitely. An output on the cli would be helpful for new users here.

I'd be happy to provide more info if you tell me what to look for.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Xubuntu 19.04

Howdy version: 2.5.1

boltgolt, 2019-08-02:
It could be that you need to set a specific resolution (see #51) to let the emitters fire. This has been reported before in Thinkpads. The resolution of 360x640 is wierd, i mostly see square formats here

j-m-p, 2019-08-02:
Using one of the native resolutions of 360x360 or 640x360 doesn't help firing the emitter. The square resolution has a much brighter image at least.

I've looked around in /sys/ for a manual control of the LED but no success there either.

```
v4l2-ctl -d  /dev/video2 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'GREY' (8-bit Greyscale)
		Size: Discrete 360x360
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 640x360
			Interval: Discrete 0.033s (30.000 fps)
```
```
v4l2-ctl -d  /dev/video2 --all
Driver Info:
	Driver name      : uvcvideo
	Card type        : Integrated Camera: Integrated I
	Bus info         : usb-0000:05:00.4-2.1
	Driver version   : 5.0.15
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 8: ok)
Format Video Capture:
	Width/Height      : 360/360
	Pixel Format      : 'GREY' (8-bit Greyscale)
	Field             : None
	Bytes per Line    : 360
	Size Image        : 129600
	Colorspace        : sRGB
	Transfer Function : Default (maps to sRGB)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Full Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 360, Height 360
	Default     : Left 0, Top 0, Width 360, Height 360
	Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 360, Height 360, Flags: 
Selection: crop_bounds, Left 0, Top 0, Width 360, Height 360, Flags: 
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 15.000 (15/1)
	Read buffers     : 0
```
```
v4l2-ctl -d  /dev/video3 --all
Driver Info:
	Driver name      : uvcvideo
	Card type        : Integrated Camera: Integrated I
	Bus info         : usb-0000:05:00.4-2.1
	Driver version   : 5.0.15
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04a00000
		Metadata Capture
		Streaming
		Extended Pix Format
Priority: 2
```

boltgolt, 2019-08-02:
I don't think even windows has control over the emitters, as far as i know they just fire when the camera is active

There's nothing i know that could fix this right now unfortunately. You could switch to using the normal webcam and use Howdy that way, but it would be less secure of course

j-m-p, 2019-08-02:
Well thanks for clearing that up. For now it's working with IR and without the emitters. Less secure probably, same as with the normal camera.

Great project!

closed 2019-08-02 by j-m-p

j-m-p, 2019-11-11:
Solution in #269. IR flasher need to be turned on for this camera model.