boltgolt/howdy #241: GStreamer warnings on Archlinux

acelot, 2019-08-07:
These messages come out every time when howdy scans my face:
```
[ WARN:0] global /build/opencv/src/opencv-4.1.1/modules/videoio/src/cap_gstreamer.cpp (1756) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global /build/opencv/src/opencv-4.1.1/modules/videoio/src/cap_gstreamer.cpp (886) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /build/opencv/src/opencv-4.1.1/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
```

My environment:
```bash
# uname -a
Linux t480s 5.2.6-arch1-1-ARCH #1 SMP PREEMPT Sun Aug 4 14:58:49 UTC 2019 x86_64 GNU/Linux

Packages versions:
extra/gstreamer 1.16.0-2 (1.9 MiB 18.0 MiB) (Installed)
extra/opencv 4.1.1-1 (30.1 MiB 174.9 MiB) (Installed)
aur/howdy 2.5.0-1 (+5 0.66%) (Out-of-date 2019-05-26) (Installed)
```

lwilletts, 2019-08-11:
Can confirm this report.

```
[ WARN:0] global /build/opencv/src/opencv-4.1.1/modules/videoio/src/cap_gstreamer.cpp (1756) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global /build/opencv/src/opencv-4.1.1/modules/videoio/src/cap_gstreamer.cpp (886) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /build/opencv/src/opencv-4.1.1/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
```

Technical27, 2019-08-31:
I also can confirm this

Slept-Warrior, 2019-09-10:
Can confirm on the 2.5.1 version installed by makepkg the PKGBUILD from this git repo
- System versions
`Manjaro XFCE 18.0.4
Linux mibrony-pc 4.19.69-1-MANJARO #1 SMP PREEMPT Thu Aug 29 08:51:46 UTC 2019 x86_64 GNU/Linux`
- Packages versions
`extra/gstreamer 1.16.0-2 (1.9 MiB 18.0 MiB) (Installed)
extra/opencv 4.1.1-1 (30.1 MiB 174.9 MiB) (Installed)
aur/howdy 2.5.0-1 (+6 1.12%) (Out-of-date 2019-05-26) (Installed: 2.5.1-1)
`
Anyone who confirmed this with same enviorment like acelot's or mine should use the +1 emoji reaction rather than reply a new comment.
If you really want to express youself, at least post the info of your environment like above so you're contributing

boltgolt, 2019-09-16:
@a3v01d please read the comment above yours

---

I can't find much about it online, but it seems to be an issue with the library OpenCV uses. There's not much i can do except hope that it gets patched soon, i'm not even sure where to report this

Slept-Warrior, 2019-09-17:
@boltgolt According to [opencv issue #14535](https://github.com/opencv/opencv/issues/14535), OpenCV has an API that control loglevel, but it's not exposed to python(cv2) yet. And the opencv binary in the official arch repo package is build on WARN level (which is by default).
But it can be set as environment variables:
`export OPENCV_LOG_LEVEL=ERROR`
This in /etc/enviorment or .profile or .(whatever shell)rc will make the warning disappear

Slept-Warrior, 2019-09-26:
I've update the [Howdy archwiki page](https://wiki.archlinux.org/index.php/Howdy#GStreamer_warnings_in_shell). I tried my best but English is not my mother tongue. I'm semi-confident but it'll be fine if you want to do some editting :P 

And I think this issue can be closed since there's nothing more can be done other than a temporary workaround. I'll comment on this issue again when the logging API get proted to the python cv2 and let you noticed.

Howdy is an amazing project, keep up the good work! :)

boltgolt, 2019-11-11:
Closing, as suggested by @Slept-Warrior

closed 2019-11-11 by boltgolt

Sobuj53, 2019-11-14:
@Slept-Warrior 
Can you please elaborate a bit more how you added `export OPENCV_LOG_LEVEL=ERROR` in `/etc/enviorment `?
I'm facing the same error but howdy is working OK. I tried `sudo nano /etc/enviorment ` then added `export OPENCV_LOG_LEVEL=ERROR` line but still when I run `sudo howdy test` it gives me the error. I think I didn't do it in the right way.

Slept-Warrior, 2019-11-18:
@Sobuj53  it's a environment variables which means it will only apply the next time that file is used.
if you're editing ~/.bashrc, it'll work after a new shell is opened. Thus, you need to logout&login or reboot to make /etc/environment applied globally