boltgolt/howdy #242: Howdy throws numpy import error on running

HEXcube, 2019-08-16:
Running `sudo howdy add` doesn't work and throws this error:

```
ImportError: numpy.core.multiarray failed to import
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/lib/security/howdy/cli/add.py", line 10, in <module>
    import cv2
  File "/usr/local/lib/python3.6/dist-packages/cv2/__init__.py", line 3, in <module>
    from .cv2 import *
ImportError: numpy.core.multiarray failed to import
```

And since no model was ever added, howdy fails everwhere.

----
Linux distribution: Ubuntu 18.04 LTS
Howdy version: 2.5.1 (installed from PPA)

HEXcube, 2019-08-16:
I found [this](https://github.com/theyosh/TerrariumPI/wiki/Known-issues#python3-numpy-import-error) and figured it's an issue with numpy version mismatch. I ran this command:

    sudo pip3 install --upgrade numpy

And then `sudo howdy add` worked! :smiley:  Now howdy works fine.

HEXcube, 2019-08-17:
numpy is now version 1.17.0 as per `pip show numpy`. It used to be 1.16 and then 1.15 when howdy was failing.

boltgolt, 2019-11-11:
Only looking at this now, thanks for figuring it out and thanks to @rushabh-v for adding an upgrade step in the installer!

closed 2019-11-11 by boltgolt