boltgolt/howdy #243: Unknown error 1 when manully lock the screen

longgangfan, 2019-08-20:
_Please describe the issue in as much detail as possible, including any errors and traces_
As the guide, I installed howdy successfully, not errors reported.
Case 1: sudo -i or sudo <whatever>
howdy works!
Case 2: restart system
howdy works!
Case 3: Ctrl+Alt+L, lock the screen
errors reports: showing Unknown errors: 1  in the login frame.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): ubuntu 16.04

Howdy version: 2.5.1

longgangfan, 2019-08-20:
I guess it is about the permission. Under the locked screen situation, it seems an regular user login, but when new boot it seems something like sudo.

longgangfan, 2019-08-20:
as https://github.com/boltgolt/howdy/issues/238#issuecomment-517842301
sudo chmod o+x  /lib/security/howdy/dlib-data/
soved the problem

boltgolt, 2019-08-20:
Thanks for figuring it out!

longgangfan, 2019-08-20:
But I found another problem. 
situation: when I sit before my computer for hours, then I want to leave. So I press Ctrl+Alt+L and locked the screen. But the problem came. Because almost the same time howdy works, and my face is recognized before I can get far from the camera. And the authority is ready in the login frame. So when I leave, anyone who clicks the "Unlock" will enter my desktop. Maybe, it can be improved. @boltgolt

boltgolt, 2019-09-16:
Unfortunately this is quite hard within the PAM system Howdy has to work in, it can be done but i will close it for now

closed 2019-09-16 by boltgolt

HZ89, 2020-06-23:
when enable ubuntu 20.04 lts fingerprint login,  got this error too in locked screen

boltgolt, 2020-06-23:
Please open a new issue for that, is off topic for this thread