boltgolt/howdy #244: `su` Reports `No face model known`

r351574nc3, 2019-08-24:
When I run `su` (see below), 
```
~ ❯❯❯ su
No face model known
```

Also, when I attempt to login after screen-lock, I see the camera activate, but I cannot tell if it ever actually logs me in using the camera.
----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): `Linux behemoth 5.2.0-kali2-amd64 #1 SMP Debian 5.2.9-2kali1 (2019-08-22) x86_64 GNU/Linux`

Howdy version: 2.5.1

tester1969, 2019-09-22:
copy your "username".dat file as root.dat file on the same directory

boltgolt, 2019-11-09:
The issue here is that Howdy is always executed as root by the PAM, and tries to determine the running user by environment variable. This can fail an return the root user, thank for the workaround @tester1969!

closed 2019-11-09 by boltgolt

Z-SHOCK-E, 2021-02-09:
type sudo su.Will work

bjmd, 2021-04-15:
> type sudo su.Will work

So I have come across this as I have hit the same problem and `sudo su -` doesn't work, weirdly. At least, it doesn't on OpenSuse. In Arch and Fedora when I run `sudo howdy add` I get a model created for my user and it works everywhere. But for some reason in OpenSUSE which I am testing at the moment, the face model for my user only works for gdm. I add it to polkit and sudo but I get the `No face model known` error, unless I symlink a root.dat to my username. I then get a confirmation for `Identified face as root`