boltgolt/howdy #245: Can't install howdy on PopOS 19.04

nekdolan, 2019-08-27:
I get this error when trying to install Howdy from the ubuntu repo:
```
>>> Installing OpenCV

Traceback (most recent call last):
  File "/var/lib/dpkg/info/howdy.postinst", line 189, in <module>
    handleStatus(subprocess.call(["pip3", "install", "--no-cache-dir", "opencv-python"]))
  File "/usr/lib/python3.7/subprocess.py", line 323, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.7/subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.7/subprocess.py", line 1522, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'pip3': 'pip3'
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.8.5-2) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

A few weeks ago I managed to install Howdy on this system without any issue, but not now.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): PopOS 19.04

Howdy version: stable

boltgolt, 2019-09-16:
Does the root user have access to pip3?

nekdolan, 2019-09-16:
I managed to install howdy somehow. Not sure what the issue was.

closed 2019-09-16 by nekdolan