boltgolt/howdy #248: add command does not terminate

dsjstc, 2019-09-01:
The `howdy add` command does not terminate -- it activates the emitters, presents the instruction "Please look straight into the camera", and remains in that state.  I've had one successful add in my current state (fresh Pop OS install), but cannot replicate.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Pop OS and Manjaro on  Lenovo X1 Extreme
Howdy version: Pop package 2.5.1

YOUR-WORST-TACO, 2019-09-04:
I am also having this issue on Kde-Neon (basically fresh), I did have to install the ppa version of opencv, the one installed by pip3 caused segmentation faults.

Also of note is that when I run `sudo howdy test` the IR camera will flash once and then the console will state: "Closing window" with no errors as seen below:
```
~ t@v sudo howdy test

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

Closing window
```

When running Cheese, I did have to change resolution for the IR camera to 400x400, which makes the image actually visible, however I modified the frame_width and frame_height inside the howdy config to no avail.
howdy config:
```
[core]
detection_notice = false
no_confirmation = false
suppress_unknown = false
ignore_ssh = true
ignore_closed_lid = true

disabled = false

use_cnn = false

[video]
certainty = 3.5
timeout = 4
device_path = /dev/v4l/by-path/pci-0000:00:14.0-usb-0:5:1.0-video-index0
max_height = 320

frame_width = 400
frame_height = 400

dark_threshold = 50
recording_plugin = opencv

device_format = v4l2
force_mjpeg = true

exposure = -1

[debug]
end_report = false
```

I am running this on a Thinkpad t480s on KDE - xorg
name -l: Linux v 5.0.0-27-generic 28~18.04.1-Ubuntu SMP

boltgolt, 2019-09-16:
The issue here is the same we had some time ago with face verification: Howdy ignores frames that are too dark, but calls [continue](https://github.com/boltgolt/howdy/blob/master/src/cli/add.py#L157) before the frame counter is incremented, thus never allowing a timeout to be reached. A workaround for this would be to set `dark_threshold` to 100 in the config, completely disabling this detection line.

---

@YOUR-WORST-TACO the only time "Closing window" should be printed is on `KeyboardInterrupt`, so pressing ctrl + C. Really no clue why it's thrown there for you

YOUR-WORST-TACO, 2019-09-16:
@boltgolt I have recently installed arch on the system, I will attempt installing howdy again to see if it occurs for me there as well

closed 2020-06-22 by boltgolt