boltgolt/howdy #249: How do I enable GPU processing?

nhusby, 2019-09-04:
I have a GTX 1050, and I would like to enable GPU processing.  The configs allude to using the GPU for the use_cnn option.  Is there a way to assert that the GPU is doing the processing?  

I enabled use_cnn, but `time howdy test` took about 5 seconds (compared to an average 1.35), so I'm not sure if it is using the GPU or not.  

Thanks

nhusby, 2019-09-04:
Also, just an FYI, this works great with my Creative "Live! Cam Socialize HD" webcam, which has no IR illumination, but the room has decent lighting.

boltgolt, 2019-09-16:
GPU usage is dependent on the OpenCV installation, if it was compiled and installed for the GPU then Howdy will use it as well. Right now, `use_cnn` only enables the more accurate detection mode, but that accuracy does come at the price of longer compute times.

ZacharyWu, 2021-03-27:
Hi, there. I have an eGPU RTX2070 attached. It works great on my laptop Xps 9575. But I want to use CPU by default. May I know how to do that? Thanks! You know, sometimes, I just carry my laptop only, but every time, it asks me the connect GPU to enable the support for howdy. That's maybe my ubuntu is configured with GPU support before. I would greatly appreciate you if there is any idea on this. Thanks again!

boltgolt, 2021-04-02:
@ZacharyWu It's probably easiest to reinstall howdy without the GPU connected if you're using Ubuntu. For other distros you need to recompile dlib without the GPU

CircleOnCircles, 2021-09-11:
How to recompile howdy? I have rtx3090 with Logitech brio. The current cnn speed is impractical.

CircleOnCircles, 2021-09-11:
> How to recompile howdy? I looked thru wiki and readme. I have rtx3090 with Logitech brio. The current cnn speed is impractical. I want some guide

after a bit of research, I think we need to install dlib to system python3. this is [an old reference](https://gist.github.com/flavindias/0d1f8a1e1692572493d7437335705a33) to install