boltgolt/howdy #252: Installation failure with "No such file or directory: 'pip3': 'pip3'"

rdeepak25, 2019-09-16:
Full error message

>>> Upgrading pip to the latest version

Traceback (most recent call last):
  File "/var/lib/dpkg/info/howdy.postinst", line 110, in <module>
    handleStatus(sc(["pip3", "install", "--upgrade", "pip"]))
  File "/usr/lib/python3.7/subprocess.py", line 323, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.7/subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.7/subprocess.py", line 1522, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'pip3': 'pip3'
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for libc-bin (2.29-0ubuntu2) ...
Processing triggers for man-db (2.8.5-2) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)

AradielSabathius, 2019-11-05:
Reinstall python 3 pip

boltgolt, 2019-11-11:
Although this issue was opened earlier, more discussion has happened at issue #277, closing this one as a duplicate.

closed 2019-11-11 by boltgolt