boltgolt/howdy #253: error installing on arch (using the yay helper)

ANDROID2468, 2019-09-17:
_Please describe the issue in as much detail as possible, including any errors and traces_
This is the output from the terminal: 

Successfully installed face-recognition-1.2.3
The --yes options to dlib's setup.py don't do anything since all these options 
are on by default.  So --yes has been removed.  Do not give it to setup.py.
==> ERROR: A failure occurred in package().
    Aborting...
Error making: howdy

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):arch

Howdy version:

boltgolt, 2019-09-18:
The only error line 
```
 ==> ERROR: A failure occurred in package().
```
Doesn't say a lot, are there no other obvious problems or errors?

ANDROID2468, 2019-09-18:
nope, but here is my  terminal output when running the command
(the command I used: yay -S howdy 2>&1 | tee log.txt)
[log.txt](https://github.com/boltgolt/howdy/files/3628663/log.txt)

pyro57, 2019-09-19:
Can confirm, even building the package from the AUR the old fashioned way (without yay) it gives the same error, see screenshot.  When googling I found this, but I'm not sure if it has anything to do with this issue.
https://bbs.archlinux.org/viewtopic.php?id=219832

[image]

pyro57, 2019-09-19:
Figured it out, the PKGBUILD file calls the setup.py for that, and passes it the --yes arugmet.  run yay -S howdy --editmenu and whe it asks you to edit PKGBUILD files select 1, type the name of the editor you want to use (for me nanao) then search for the --yes option and delete it.  however now it complains about USE_AVX_INSTRUCTIONS being an invalid command, still trouble shooting.

pyro57, 2019-09-19:
can confirm, steps to install:

1. run yay -S howdy --editmenu.
2. when it asks you to edit pkgbuild select 1 howdy
3. use what ever editor you want
4. find --yes and delete it and the USE_AVX_INSTRUCTIONS 
5. continue as normal

Maintainer - update your pkgbuild to reflect these changes thank you.

closed 2021-04-02 by boltgolt