boltgolt/howdy #254: Getting OpenCV error

singhanurag9080, 2019-09-18:
Adding face model for the user singhanurag9080
Enter a label for this new model [Initial model] (max 24 characters): anurag
VIDEOIO ERROR: V4L2: Could not obtain specifics of capture window.
/dev/v4l/by-path/pci-0000:00:14.0-usb-0:9:1.0-video-index1 does not support memory mapping

Please look straight into the camera
Traceback (most recent call last):
  **File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 147, in <module>
    gsframe = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.1.1) /io/opencv/modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'**

boltgolt, 2019-09-19:
More info than just an error trace is required, try putting a little more effort into your issue. The issue template did ask you for a howdy and distro version. Either way the last part you had highlighted isn't the real error, this is:

```
VIDEOIO ERROR: V4L2: Could not obtain specifics of capture window.
/dev/v4l/by-path/pci-0000:00:14.0-usb-0:9:1.0-video-index1 does not support memory mapping
```

What camera are you trying to use?

singhanurag9080, 2019-09-19:
Sorry I am new to GitHub,
I made some changes in the config file(replace video-index1 with video-index1)
**# The path of the device to capture frames from
# Should be set automatically by an installer if your distro has one
device_path = /dev/v4l/by-path/pci-0000:00:14.0-usb-0:9:1.0-video-index0**

My device and path are: 
**HP HD Camera: HP HD Camera (usb-0000:00:14.0-9):
	/dev/video0
	/dev/video1

crw-rw----+  1 root            video    81,     1 Sep 19 14:25 video0
crw-rw----+  1 root            video    81,     2 Sep 19 14:25 video1**

I am able to add a face, everything's working fine but, on running test I am getting the following error : 

**X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x2c00010**
Got any views on it,

KhinShoon, 2019-09-30:
> Sorry I am new to GitHub,
> I made some changes in the config file(replace video-index1 with video-index1)
> Now, howdy add is not giving any error but is stuck at capturing the face.
> 
> ```
>  # The path of the device to capture frames from
>  #254 Should be set automatically by an installer if your distro has one
> device_path = /dev/v4l/by-path/pci-0000:00:14.0-usb-0:9:1.0-video-index0
> ```
> 
> My device and path are:
> 
> ```
> HP HD Camera: HP HD Camera (usb-0000:00:14.0-9):
> 	/dev/video0
> 	/dev/video1
> 
> crw-rw----+  1 root            video    81,     1 Sep 19 14:25 video0
> crw-rw----+  1 root            video    81,     2 Sep 19 14:25 video1
> ```
> 
> On running the test I am getting the following error :
> 
> ```
> X Error: BadDrawable (invalid Pixmap or Window parameter) 9
>   Major opcode: 62 (X_CopyArea)
>   Resource id:  0x2c00010
> ```
> 
> Got any views on it,

I also got the same error with replacing the video index. Any way to fix this problem?

KhinShoon, 2019-09-30:
@boltgolt hope you can review on it.

KhinShoon, 2019-09-30:
> Sorry I am new to GitHub,
> I made some changes in the config file(replace video-index1 with video-index1)
> Now, howdy add is not giving any error but is stuck at capturing the face.
>
> ```
>  # The path [...]

@singhanurag9080 I disabled howdy then re-do 'sudo howdy add' instead of testing it. Make sure you cleared all the face IDs you have added before. Then, you enable the howdy back and it works perfectly for mine.

boltgolt, 2019-11-11:
The `X Error: BadDrawable` part is a duplicate issue, you can find a workaround at #60

closed 2019-11-11 by boltgolt