boltgolt/howdy #255: Install fails on Ubuntu 19.04 / howdy 2.5.1

anarchist912, 2019-09-19:
After the build is completed `sudo apt install howdy` fails:

```
>>> Cleaning up dlib

Temporary dlib files removed

>>> Installing OpenCV

Traceback (most recent call last):
  File "/var/lib/dpkg/info/howdy.postinst", line 189, in <module>
    handleStatus(subprocess.call(["pip3", "install", "--no-cache-dir", "opencv-python"]))
  File "/usr/lib/python3.7/subprocess.py", line 323, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.7/subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "/usr/lib/python3.7/subprocess.py", line 1522, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'pip3': 'pip3'
```
pip3 however is installed. I can manually complete the step:
```
pip3 install --no-cache-dir opencv-python
Requirement already satisfied: opencv-python in /usr/local/lib/python3.7/dist-packages (4.1.0.25)
Requirement already satisfied: numpy>=1.14.5 in /usr/local/lib/python3.7/dist-packages (from opencv-python) (1.16.3)
```
But howdy is not available from the shell and can't be enabled. My DELL XPS 13 inbuilt IR-Camera however can be tested by running `sudo python3 cli.py test` and I can add a face model.

tester1969, 2019-09-27:
Create a link
[root@oc3588317528 models]# ls -l /bin/howdy
lrwxrwxrwx. 1 root root 32 Sep 22 23:42 /bin/howdy -> /usr/lib64/security/howdy/cli.py

i guess everything is there in /usr/lib(64?)/security/howdy
[root@oc3588317528 pam.d]# cd /usr/lib64/security/howdy
[root@oc3588317528 howdy]# ls -l
total 28
drwxr-xr-x. 3 root root 4096 Sep 22 23:42 cli
-rwxr-xr-x. 1 root root 3314 Nov  9  2018 cli.py
-rw-r--r--. 1 root root 5410 Nov  9  2018 compare.py
-rw-r--r--. 1 root root 1762 Sep 23 00:00 config.ini
drwxr-xr-x. 2 root root 4096 Sep 26 15:52 models
-rw-r--r--. 1 root root 2611 Nov  9  2018 pam.py

then add in /etc/pam.d/sudo
auth       sufficient   pam_python.so /usr/lib64/security/howdy/pam.py

test if sudo works. (make also sure that pam_python.so is available.

anarchist912, 2019-09-27:
@tester1969 
So in my case:
```
sudo ln -s /lib/security/howdy/cli.py /bin/howdy
which howdy
/bin/howdy
```
Also added to /etc/pam.d/
```
auth sufficient pam_python.so /lib/security/howdy/pam.py
```
But where should pam_python.so live?

```
ls -1 /lib/security/howdy
cli
cli.py
compare.py
config.ini
dlib-data
models
pam-config
pam.py
recorders
```

closed 2019-09-27 by anarchist912