boltgolt/howdy #261: Install problem on Lenovo X390 Yoga

pczekalski, 2019-10-07:
First of all, thanks for your effort bringing new features to the Linux!

While installing for Lenovo Yoga X390 I've got problem transmitters not triggering at all on any of four configurations. Only RGB camera seems to work (got white LED presenting it on) but on IR beam. Any hints on how to make it work are highly appreciated.

pczekalski, 2019-10-07:
Some more information:
`v4l2-ctl -d /dev/video2 --all
Driver Info:
	Driver name      : uvcvideo
	Card type        : Integrated Camera: Integrated I
	Bus info         : usb-0000:00:14.0-8
	Driver version   : 5.0.21
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 8: ok)
Format Video Capture:
	Width/Height      : 640/360
	Pixel Format      : 'GREY' (8-bit Greyscale)
	Field             : None
	Bytes per Line    : 640
	Size Image        : 230400
	Colorspace        : sRGB
	Transfer Function : Default (maps to sRGB)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Full Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 640, Height 360
	Default     : Left 0, Top 0, Width 640, Height 360
	Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 640, Height 360, Flags: 
Selection: crop_bounds, Left 0, Top 0, Width 640, Height 360, Flags: 
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0
`

pczekalski, 2019-10-08:
I've got some progress, both cameras work /dev/video0 (RGB) and /dev/video2 (IR) using `fswebcam.` IR camera gives 640x360 (assume it is correct) but still, no IR blasters working.

DavidM42, 2019-10-23:
Got the same problem on my Lenovo Yoga S740

PetePriority, 2019-11-01:
@pczekalski This might be related to #269. Could you try if the workaround https://github.com/boltgolt/howdy/issues/269#issuecomment-548105971 works for you?

pczekalski, 2019-11-04:
Checked, still no luck.
I've set up Windows guest (Ubuntu 19.04 host) on VirtualBox and under the Windows 10 guest, Windows Hello works well. Wireshark installed on the host side but I'm stuck with configuring it as there is no "USB" interface in the options. Any hints on where to start configuration for sniffing?
Thanks in advance.

PetePriority, 2019-11-04:
> Checked, still no luck.
> I've set up Windows guest (Ubuntu 19.04 host) on VirtualBox and under the Windows 10 guest, Windows Hello works well. Wireshark installed on the host side but I'm stuck with [...]

You need to load the `usbmon` kernel module: `modprobe usbmon`

pczekalski, 2019-11-04:
Thx,

OK - did a dump. In my case, it is a **usbmon1** interface. However, I still don't get it if usbmon1 is a single device or a bunch of them connected to the bus. If suspect the second case as it seems to capture many devices: 1.4.0, 1.4.1, 1.5.0 and 1.5.2 (as showed up when loaded to wireshark). Any hints, how to filter to the particular device?

boltgolt, 2019-11-11:
Both this and #269 are about Lenovo cameras and debugging the issue with USB messages, so marking this as a duplicate to keep the discussion in one place

closed 2019-11-11 by boltgolt