boltgolt/howdy #262: AUR package is broken, mind if I maintain it?

Raymo111, 2019-10-13:


boltgolt, 2019-10-17:
Sure, if you have the time to do so. Do you have any prior experience in package maintenance?

Raymo111, 2019-10-17:
No, but I'm pretty active and willing to spend time doing so. I've drafted a PKGBUILD already: https://gist.github.com/Raymo111/1c795b57e3fc71922b5e43ee13e92319

Raymo111, 2019-10-24:
@boltgolt Can you make me a maintainer to the AUR package so I can upload the new PKGBUILD?

aouerfelli, 2019-10-24:
For what it's worth, @kageurufu also offered to maintain this package in #186.

kageurufu, 2019-10-24:
I would still be happy to take over maintenance as well, or work together with @Raymo111

Raymo111, 2019-10-24:
@kageurufu Same here. :)

xuanruiqi, 2019-10-25:
I would be very happy to maintain it. I am already maintainer of 9 AUR packages ([here](https://aur.archlinux.org/packages/?SeB=m&K=xuanruiqi)) and don't mind maintaining this one as well.

Also, just saying, we probably should not be cloning and installing things using pip inside the PKGBUILD... I can (and am happy to) look into ways to fix that.

Raymo111, 2019-10-25:
Ok, whoever maintains it, I suggest you take a look at the way I import dependencies: https://gist.github.com/Raymo111/1c795b57e3fc71922b5e43ee13e92319

kageurufu, 2019-10-25:
@xuanruiqi #186 already fixes that, and also pre-installs the trained models to `/usr/lib/security/howdy/dlib-data/`

xuanruiqi, 2019-10-25:
OK, thanks for the note!

noomly, 2019-11-04:
Is there some update on this issue? It looks like the AUR package is still outdated as of March and the installation fails.

kageurufu, 2019-11-04:
@noomly feel free to pull down the PKGBUILDs I posted in #186, they're fully functional. I'm working on getting them on the aur

noomly, 2019-11-04:
I used the PKGBUILD from https://github.com/boltgolt/howdy/tree/master/archlinux and it kind of worked (I had to do a few manual things to get it to work but it works on my computer now).

It just seems that there is not a lack of people willing to maintain this package and are just waiting for @boltgolt to give the necessary permissions, so I wanted to ping as nothing appears to be keeping him from doing so.

boltgolt, 2019-11-11:
The delay is that i can't contact the original maintainer of the package on AUR, so i can't add anyone as a co-maintainer. I've send a orphan request to the AUR team.

boltgolt, 2019-11-14:
@Raymo111 @kageurufu @xuanruiqi I'm now the maintainer of the arch package, what are your names on AUR?

kageurufu, 2019-11-14:
I am `kageurufu` there as well

Raymo111, 2019-11-14:
I am `Raymo111` there as well.

xuanruiqi, 2019-11-15:
I'm `xuanruiqi` there as well.

boltgolt, 2019-11-15:
Easier than expected, you're all added as co-maintainers. I trust that you all can create the best PKGBUILD possible with your combined effords

kageurufu, 2019-11-15:
@Raymo111 @xuanruiqi My PKGBUILD from #186 is still functional, builds against Python3.8 just fine, and automates downloading the dlib models as part of packaging instead of requiring downloading them manually. The package builds in a clean chroot, so it has no missing dependencies, and the built packages work on my machine.

@boltgolt would you be willing to give commit rights on this repo, so I/we can maintain the archlinux/ packaging folder in the repo? Otherwise, we can just send pull requests for any changes

Raymo111, 2019-11-15:
@kageurufu Take a look at how I import dependencies. A comment on the AUR mentioned that not only was the original PKGbuild broken but the method for deps was off. https://gist.github.com/Raymo111/1c795b57e3fc71922b5e43ee13e92319

kageurufu, 2019-11-15:
`python-pam` you have there isn't the right package, its a python library for authenticating using pam. `pam-python` is a pam plugin allowing pam to call out to python to check authentication

I also think including the dlib models in the packaging process is valuable to the user, as it makes the package usage more intuitive

Raymo111, 2019-11-15:
Alright, @boltgolt if you'll let @kageurufu and I know whether you'll give us commit rights to this repo or stick to PRs, then we can close this issue. I think it's pretty much been resolved.

boltgolt, 2019-11-19:
I'd like to stick to PRs for now, I've _finally_ merged the PR

Thanks for all the amazing work and putting up with my slow response times everyone!

closed 2019-11-19 by boltgolt