boltgolt/howdy #263: Howdy CLI Not Starting on Ubuntu 19.04

RD1K, 2019-10-13:
I just installed Howdy using the commands provided in the README file here. It installed fine; if I enter `sudo apt install howdy` into the terminal it says `howdy is already the newest version (2.5.1)`. However, when I try to add a face with `sudo howdy add`, or even if I just enter in `howdy` or `sudo howdy config`, I get this output: `howdy: command not found`. Why might this be?
----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu 19.04 Desktop

Howdy version: 2.5.1

boltgolt, 2019-10-17:
Weird, what does `whereis howdy` output for you?

RD1K, 2019-10-17:
It outputs "howdy: /use/share/man/man1/howdy.1.gz"

Bound3R, 2019-10-23:
I have the same problem, I really wanted to try this package. :cry:

atharvaunde59, 2019-10-24:
I'm also facing same issue on Ubuntu 19.10

root@atharva-Inspiron-5559:/usr/bin# sudo howdy add
sudo: howdy: command not found
root@atharva-Inspiron-5559:/usr/bin# whereis howdy
howdy: /usr/share/man/man1/howdy.1.gz

Bound3R, 2019-10-25:
Hello, I followed these steps to install it manually and this method worked for me you can see the thread of issue [#256 ](https://github.com/boltgolt/howdy/issues/265) 

on resumen i run this comands, maybe need sudo.
```
apt remove howdy
pip3 install --upgrade pip
sudo dpkg --install howdy_2.5.1.deb
```

RD1K, 2019-10-27:
@Bound3R Where did you get the .deb file from?

Bound3R, 2019-10-28:
> @Bound3R Where did you get the .deb file from?

Directly from the [releases page](https://github.com/boltgolt/howdy/releases), to be more specific use version [_2.5.1_ ](https://github.com/boltgolt/howdy/releases/download/v2.5.1/howdy_2.5.1.deb).

RD1K, 2019-11-04:
I switched distros to Manjaro a few days ago, so I will no longer be able to test any solutions. Therefore, I'm closing this thread.

closed 2019-11-04 by RD1K

matifali, 2020-09-02:
> apt remove howdy
> pip3 install --upgrade pip
> sudo dpkg --install howdy_2.5.1.deb

This worked for me, exact commands I used were
```
sudo apt remove howdy
sudo pip3 install --upgrade pip
sudo dpkg --install howdy_2.6.0.deb
```

genidma, 2020-09-28:
> sudo dpkg --install howdy_2.6.0.deb

That's not working atm.

telefonosuci, 2021-08-03:
> Hello, I followed these steps to install it manually and this method worked for me you can see the thread of issue [#256 ](https://github.com/boltgolt/howdy/issues/265)
>
> on resumen i run this [...]

@Bound3R Where can I download the .deb file ?

boltgolt, 2021-08-03:
@telefonosuci https://github.com/boltgolt/howdy/releases/tag/v2.6.1