boltgolt/howdy #264: Add false color to IR emitter for enhanced security.

papermache, 2019-10-15:
If Howdy authenticated by false color from a heat signature in addition to the face model it would be harder to spoof with a photo.

boltgolt, 2019-11-11:
Although i like the idea, Howdy is made to be a replacement for laptops that support Windows Hello and which normally do not have a thermal imaging camera. Closing as out of scope, but only because the required hardware just isn't there.

closed 2019-11-11 by boltgolt

papermache, 2019-11-11:
Does false color require special hardware or is the IR emitter sufficient?

boltgolt, 2019-11-11:
In this case the image returned from the camera is only in black and white, there is no exposed thermal data

papermache, 2019-11-11:
I think if you assign colors to different shades of grey received by heat
from the IR emitter false color adds in so no special hardware is required.

https://github.com/fk128/falseColor

boltgolt, 2019-11-11:
That's not reliable though, a well printed picture on a piece of paper will have more or less the same IR image as a real human. We're shining IR light on the subject through the emitters and then measuring how much of the light gets reflected back, it has very little to do with how much IR radiation the subject itself is emitting

papermache, 2019-11-11:
Can the colors therefore be assigned by the reflected light instead of heat?

boltgolt, 2019-11-11:
We could, but it would not be effective in determining if a subject is alive or not, and thus not help prevent picture spoofing. We would be adding false colors to both real humans and pictures and the difference would not be noticeable.

papermache, 2019-11-11:
OK, I get it. Could basic ray tracing be an alternative to reduce spoofing?

https://www.scratchapixel.com/lessons/3d-basic-rendering/introduction-to-ray-tracing/implementing-the-raytracing-algorithm

papermache, 2019-11-11:
Here is another small raytracing program. There are quite a few repos.

https://github.com/embedded2016/raytracing