boltgolt/howdy #265: Error while installing Ubuntu 19.04

pontussalenbo, 2019-10-16:
Unpacking howdy (2.5.1) ...
Setting up howdy (2.5.1) ...

>>> Upgrading pip to the latest version

Traceback (most recent call last):
  File "/usr/bin/pip3", line 11, in <module>
    sys.exit(main())
TypeError: 'module' object is not callable
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.8.5-2) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)

Linux distribution (if applicable): Ubuntu 19.04

Howdy version: 2.5.1

Any ideas on how to troubleshoot? Have tried to install serveral times and same error each and every time.

leighleighleigh, 2019-10-16:
I had the same problem, I found that it installed by downloading the deb file (under the Debian section) and installing with `sudo dpkg --install howdy_2.5.1.deb`. 
Still debugging why the official Ubuntu install method won't work...

JavaWantaBe, 2019-10-22:
Either official repo or .deb file installation fails on 19.10.

boltgolt, 2019-10-22:
What happens when you run `pip3 install --upgrade pip`?

jmlago91, 2019-10-23:
Same error here. Thinkpad T495s
Nothing strange after running pip3 upgrade:
`Requirement already up-to-date: pip in /usr/local/lib/python3.7/dist-packages (19.3.1)`
After installing with .deb package, IR camera never turns on.

TheBrokenRail, 2019-10-24:
PIP changed their entry point so ```/usr/bin/pip3``` broke.
To fix this change: ```from pip._internal import main``` to ```from pip._internal.main import main``` in ```/usr/bin/pip3```

jmlago91, 2019-10-24:
After that change installation finish, but no IR camera detected on ThinkPad T495s, that I assume is other issue. Camera: 04f2:b67c Chicony Electronics Co., Ltd

TheBrokenRail, 2019-10-24:
I have a T495s and it worked for me, the camera was ```Integrated Camera C```.

jmlago91, 2019-10-24:
Just an extra detail.
Authentication is working like a charm with the IR camera but when I reboot the computer, still asks for the password. After entering the password and hitting enter, I can see the IR camera blasting for a second. Is this the expected behaviour?
I'd rather to use only the IR camera, since I have to use my fingerprint to unlock the computer, and to enter manually a password to decrypt the partition.

Bound3R, 2019-10-25:
> I cleaned up everything and reinstalled and worked perfectly with pip3 entry point modification.
> Thank you for your help.

Thanks, work for me.

Seltonu, 2019-10-28:
Hello,

I'm having this same error installing Howdy, installing via the .deb doesn't fix it. Can somebody explain to me how to do the pip3 entry point modification? /usr/bin/pip3 is non-existent.

shenzou, 2019-11-09:
> PIP changed their entry point so `/usr/bin/pip3` broke.
> To fix this change: `from pip._internal import main` to `from pip._internal.main import main` in `/usr/bin/pip3`

Thank you, this solution worked for me !

mchassy, 2020-01-11:
Hello,
I am having the same issue. I use linuxbrew and when I check pip3 under `/home/linuxbrew/.linuxbrew/bin`
```
#!/home/linuxbrew/.linuxbrew/opt/python/bin/python3.7
# EASY-INSTALL-ENTRY-SCRIPT: 'pip==19.3.1','console_scripts','pip3'
__requires__ = 'pip==19.3.1'
import re
import sys
from pkg_resources import load_entry_point

if __name__ == '__main__':
    sys.argv[0] = re.sub(r'(-script\.pyw?|\.exe)?$', '', sys.argv[0])
    sys.exit(
        load_entry_point('pip==19.3.1', 'console_scripts', 'pip3')()
    )
```
I have something different than under `/usr/local/bin`
```
#!/usr/bin/python3
# -*- coding: utf-8 -*-
import re
import sys

from pip._internal.main import main

if __name__ == '__main__':
    sys.argv[0] = re.sub(r'(-script\.pyw?|\.exe)?$', '', sys.argv[0])
    sys.exit(main())
~                          
```

Could this be causing the problem?

mchassy, 2020-01-11:
Update -- I uninstalled linuxbrew so that the only pip3 is in /usr/local/bin. I am still getting the same error. What else could I look at?

boltgolt, 2020-01-15:
Could you give my your `whereis pip3`?

mchassy, 2020-01-15:
```
❯ whereis pip
pip: /usr/local/bin/pip3.7 /usr/local/bin/pip

```

boltgolt, 2020-03-11:
@mchassy you might be able to manually edit `control.tar.gz/./postinst` in the deb or the debian source here from github to disable the 2 pip commands on install. I think removing line 110 and 189 from the `postinst` file should do it. Don't forget to install opencv-python manually afterwards

mchassy, 2020-03-14:
@boltgolt once I clone the repo and make the changes above. How do I actually install from that local repo?

boltgolt, 2020-04-01:
Yes, use your custom installer

mchassy, 2020-04-01:
Thanks @boltgolt. Finally got this installed.

boltgolt, 2020-04-01:
Good to hear! Might catch failing pip commands gracefully in the future

mchassy, 2020-04-01:
just for info

- I changed all of the pip files as indicated above to `from pip._internal.main import main`
- I downloaded the .deb, made the mods and recreated the .deb
- I did a `sudo apt purge` and `sudo apt autoremove`
- had to do a `sudo apt --fix-broken install
- I tried using the custom installer on pop_os but that hung and failed. 
- finally did `sudo dpkg --install howdy_2.5.1.fixed.deb`

There was a final problem which was that opencv-python had not been installed for the root user (I usually manage my modules as my own user in a venv for each project. So I did a `sudo su -` and installed opencv. 

Once that was all done, I was able to use `sudo howdy add`

Could this be used to login into my laptop?

closed 2021-04-02 by boltgolt

CoderThomasB, 2021-07-11:
I used `sudo ln /usr/local/bin/pip3 /usr/bin/pip3` to create a file link and fix this issue