boltgolt/howdy #269: IR Emitters not turning on on Lenovo S740

DavidM42, 2019-10-23:
I'm sorry but I have another one of these IR Emitters don't turn on issues and I couldn't resolve it with the solutions of previous issues.
My Lenovo Yoga S740 has 4 cameras registred under `/dev/video` (0-3):
* video0 is the rgb camera which works in with howdy test but isn't IR (is called Integrated C)
* video1 does not work (also called Integrated C)
* video2 is the IR webcam which technically works with howdy test but the IR emitters don't turn on so the image is very dark (is called Integrated I)
* video3 does not work (also called Integrated I)

I read that setting different resolutions could help so I tested some resolutions but sadly no width/height I tried helped (including 640/360).

Below the debug output for video2 (IR webcam):
`v4l2-ctl -d /dev/video2 --all`
```
Driver Info:
	Driver name      : uvcvideo
	Card type        : Integrated Camera: Integrated I
	Bus info         : usb-0000:00:14.0-5
	Driver version   : 5.3.1
	Capabilities     : 0x84a00001
		Video Capture
		Metadata Capture
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps      : 0x04200001
		Video Capture
		Streaming
		Extended Pix Format
Priority: 2
Video input : 0 (Camera 8: ok)
Format Video Capture:
	Width/Height      : 640/360
	Pixel Format      : 'GREY' (8-bit Greyscale)
	Field             : None
	Bytes per Line    : 640
	Size Image        : 230400
	Colorspace        : sRGB
	Transfer Function : Default (maps to sRGB)
	YCbCr/HSV Encoding: Default (maps to ITU-R 601)
	Quantization      : Default (maps to Full Range)
	Flags             : 
Crop Capability Video Capture:
	Bounds      : Left 0, Top 0, Width 640, Height 360
	Default     : Left 0, Top 0, Width 640, Height 360
	Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 640, Height 360, Flags: 
Selection: crop_bounds, Left 0, Top 0, Width 640, Height 360, Flags: 
Streaming Parameters Video Capture:
	Capabilities     : timeperframe
	Frames per second: 30.000 (30/1)
	Read buffers     : 0
```

----

Device: Lenovo Yoga S740
Distro: Pop!OS 19.10 (Ubuntu 19.10)
Howdy version: 2.5.1 .deb installer because apt repository did not work

PetePriority, 2019-10-27:
I'm having similar issues with my Lenovo T490s. I also have 4 cameras and `/dev/video2` is the IR camera. The ` v4l2-ctl -d /dev/video2 --all` output is identical except mine has a driver version of 5.3.7 and a bus info of usb-0000:00:14.0-8.

Could you check your output of `lsusb` to see if we have the same chip?
```
# lsusb
Bus 001 Device 004: ID 04f2:b67c Chicony Electronics Co., Ltd Integrated Camera
```

DavidM42, 2019-10-27:
Yep is the same
```
# lsusb 
Bus 003 Device 002: ID 04f2:b67c Chicony Electronics Co., Ltd Integrated Camera
```

PetePriority, 2019-10-27:
I found this in the Windows 10 SDK documentation: https://docs.microsoft.com/en-us/windows-hardware/drivers/stream/uvc-extensions-1-5#22211-ir-torch-control
Judging from the output of `lsusb -d 04f2:b67c -v` the device seems to have multiple extended control units, including the microsoft extended control unit described in the article and also a realtek extended controls unit:
```
      VideoControl Interface Descriptor:
        bLength                25
        bDescriptorType        36
        bDescriptorSubtype      6 (EXTENSION_UNIT)
        bUnitID                11
        guidExtensionCode         {45b5da73-23c1-4a3d-a368-610f078c4397}
        bNumControl             0
        bNrPins                 1
        baSourceID( 0)         15
        bControlSize            0
        iExtension              0 
      VideoControl Interface Descriptor:
        bLength                27
        bDescriptorType        36
        bDescriptorSubtype      6 (EXTENSION_UNIT)
        bUnitID                 9
        guidExtensionCode         {1229a78c-47b4-4094-b0ce-db07386fb938}
        bNumControl             2
        bNrPins                 1
        baSourceID( 0)         11
        bControlSize            2
        bmControls( 0)       0x00
        bmControls( 1)       0x06
        iExtension              0 
      VideoControl Interface Descriptor:
        bLength                27
        bDescriptorType        36
        bDescriptorSubtype      6 (EXTENSION_UNIT)
        bUnitID                10
        guidExtensionCode         {0f3f95dc-2632-4c4e-92c9-a04782f43bc8}
        bNumControl             9
        bNrPins                 1
        baSourceID( 0)          9
        bControlSize            2
        bmControls( 0)       0x22
        bmControls( 1)       0x01
        iExtension              8 Microsoft Extended Controls Unit
      VideoControl Interface Descriptor:
        bLength                29
        bDescriptorType        36
        bDescriptorSubtype      6 (EXTENSION_UNIT)
        bUnitID                14
        guidExtensionCode         {26b8105a-0713-4870-979d-da79444bb68e}
        bNumControl             2
        bNrPins                 1
        baSourceID( 0)         10
        bControlSize            4
        bmControls( 0)       0x00
        bmControls( 1)       0x60
        bmControls( 2)       0x00
        bmControls( 3)       0x00
        iExtension             10 Realtek Extended Controls Unit
```
I also stumbled over this: https://linuxtv.org/downloads/v4l-dvb-apis/v4l-drivers/uvcvideo.html
I have little experience with USB drivers but haven't done anything with uvc. The command `v4l2-ctl -l -d /dev/video2` doesn't return anything.
 Maybe this is helpful to someone? I can also provide usbpcaps of a Windows 10 application (Mirametrix Glance)  turning on the camera and the IR-LED.

PetePriority, 2019-10-28:
After adding some debug prints to the uvcvideo kernel driver, I managed to figure out that the extension unit supports MSXU_CONTROL_EXPOSURE, MSXU_CONTROL_FACE_AUTHENTICATION, and MSXU_CONTROL_METADATA. The second one sounded promising.

The code below sends a UVCIOC_CTRL_QUERY ioctl to `/dev/video2`, setting the "Authentication Alternative Frame Illumination" mode on the device. This will make the IR LED flicker with each frame (when video is streamed, e.g., using VLC or `howdy test`). Unfortunately direct IR Torch control (MSXU_CONTROL_IR_TORCH) doesn't seem to be implemented/supported.

```
#include <sys/types.h>
#include <sys/stat.h>
#include <fcntl.h>
#include <errno.h>
#include <unistd.h>
#include <stdio.h>
#include <stdlib.h>
#include <string.h>

#include <linux/videodev2.h>
#include <linux/uvcvideo.h>
#include <linux/usb/video.h>

#include <sys/ioctl.h>

int main() {
    __u8 setbuffer[256] = {
        0x01, // bNumEntries
        0x03, // bEndpointAddress
        0b010 // bmControlFlags
              // Bit0: Authentication Disabled
              // Bit1: Authentication Alternative Frame Illumination
              // Bit2: Authentication Background Subtraction
              // Bit3-55: Reserved, set to zero
    };    

    struct uvc_xu_control_query set_query = {
        .unit = 10,         // bUnitID
        .selector = 0x06,   // MSXU_CONTROL_FACE_AUTHENTICATION
        .query = UVC_SET_CUR,
        .size = 9,          // only one endpoint so 1+1+7
        .data = (__u8*)&setbuffer,
    };

    int fd = open("/dev/video2", O_WRONLY);
    int result = ioctl(fd, UVCIOC_CTRL_QUERY, &set_query);
    if (result < 0) {
        printf("Error code: %d, errno: %d, error: %s\n", result, errno, strerror(errno));
    }
    close(fd);

    return 0;
}
```

DavidM42, 2019-10-28:
Nice findings. I don't have any experience on this low level c code so I can't help you develop but if you need someone to test I can surely do that on my device with some instructions on how to do it

PetePriority, 2019-10-30:
I think I found the right call. In the wireshark USB dump I found URB_CONTROL messages transmitting only 2 bytes of data:

```
bmRequestType: 0x21
bRequest: 1
wValue: 0x0e00
wIndex: 0x0e02
wLength: 2
Data: 0x0219
```
The index tells us this corresponds to unit `0x0e` (Realtek Extended Controls Unit, GUID 26b8105a-0713-4870-979d-da79444bb68e), interface number 2. The value tells us the selector `0x0e`.

I don't know what the first byte of the data does. The second byte seems to control the brightness of the IR emitter. It's maxing out at 0x19 already. I've found another call with data = `0x0264`.

The following C code would turn on the emitter:
```
__u8 setbuffer[256] = {0x02, 0x19};
struct uvc_xu_control_query set_query = {
    .unit = 0x0e,
    .selector = 0x0e,
    .query = UVC_SET_CUR,
    .size = 2,
    .data = (__u8*)&setbuffer,
};
int fd = open("/dev/video2", O_WRONLY);
ioctl(fd, UVCIOC_CTRL_QUERY, &set_query);
close(fd);
```

I've put a small tool written in C to toggle these values on https://github.com/PetePriority/chicony-ir-toggle. The repo also includes a python script that does the same. The IR emitter should turn on as soon as you start streaming video from the device. Sometimes the emitter only lights up for an instant- reopening the device seems to fix that.

@DavidM42 could you test if this works for you?

DavidM42, 2019-10-30:
@PetePriority Wow very nice. That worked for me to enable the IR emitter. I used the gcc command to build it. Still have to work out adding a new face model now

PetePriority, 2019-11-01:
Try adjusting the `dark_threshold` value in your config file. I set mine to 80. I think there's a bug related to this value though. Could you try if #276 works for you?

DavidM42, 2019-11-01:
Wow thanks!
I monkey patched your PR into the installed files on my drive and set dark_threshold to 90 and now it works

pczekalski, 2019-11-01:
I'm in similar shoes with Lenovo Yoga X390 but in my case the device is:
Bus 001 Device 005: ID 13d3:56ba IMC Networks

Assuming the only option is to "Wireshark" the communication with working Windows solution, could someone guide me to do it? Sorry for beginners question but actually never used it before.

Do you work with Wireshark on the Windows Host, or Linux host (here Ubuntu) and Windows Guest on VirtualBox and USB redirects? If the second one, let me know, please if you use Wireshark on the Host side or Guest one?

Thanks in advance,

P.

PetePriority, 2019-11-01:
@pczekalski Could you post the output of `lsusb -d 13d3:56ba -v`? 

I used wireshark on the Windows Host. It should also be possible to use it on a linux host with usb redirects to a VM. I recommend having a look at some guides/tutorials/videos on how to do it.
A first step would be to find the correct usb monitoring device that the camera is attached to. Then apply some filters to remove packets from other devices on the same bus, e.g., mice and keyboards. Another useful filter is to exclude isochronous URBs (that's the video stream) using the filter `!(usb.transfer_type == 0x00)`.

pczekalski, 2019-11-01:
>"Could you post the output of lsusb -d 13d3:56ba -v?"
Absolutely, this is where I actually started ;-).
I believe the IR camera is the one in the second part as resolution seems to be the one (640x360).
```
:~$ lsusb -d 13d3:56ba -v
 
Bus 001 Device 005: ID 13d3:56ba IMC Networks
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.01
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x13d3 IMC Networks
  idProduct          0x56ba
  bcdDevice           69.09
  iManufacturer           3
  iProduct                1
  iSerial                 2
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x04cb
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          4
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0
      iFunction               5
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      1
      iInterface              5
      VideoControl Interface Descriptor:
        bLength                13
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdUVC               1.50
        wTotalLength       0x006d
        dwClockFrequency       15.000000MHz
        bInCollection           1
        baInterfaceNr( 0)       1
      VideoControl Interface Descriptor:
        bLength                18
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0201 Camera Sensor
        bAssocTerminal          0
        iTerminal               0
        wObjectiveFocalLengthMin      0
[... 797 lines cut ...]
          Synch Type               Asynchronous
          Usage Type               Data
        wMaxPacketSize     0x0300  1x 768 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       5
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      2 Video Streaming
      bInterfaceProtocol      1
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            5
          Transfer Type            Isochronous
          Synch Type               Asynchronous
          Usage Type               Data
        wMaxPacketSize     0x0a30  2x 560 bytes
        bInterval               1
```

PetePriority, 2019-11-01:
@pczekalski Judging from the output your device seems to offer the same extension units as our cameras. Have you tried the workaround proposed in https://github.com/boltgolt/howdy/issues/269#issuecomment-548105971?

kpucynski, 2019-11-07:
I have the same issue on X1 Extreme.
```
Bus 001 Device 006: ID 04f2:b615 Chicony Electronics Co., Ltd Integrated IR Camera
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.01
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x04f2 Chicony Electronics Co., Ltd
  idProduct          0xb615 
  bcdDevice           10.13
  iManufacturer           1 SunplusIT Inc
  iProduct                2 Integrated IR Camera
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x025d
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0 
      iFunction               4 Integrated IR Camera
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      0 
      iInterface              4 Integrated IR Camera
      VideoControl Interface Descriptor:
        bLength                13
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdUVC               1.00
        wTotalLength       0x00a7
        dwClockFrequency       48.000000MHz
        bInCollection           1
        baInterfaceNr( 0)       1
      VideoControl Interface Descriptor:
        bLength                18
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0201 Camera Sensor
        bAssocTerminal          0
        iTerminal               0 
        wObjectiveFocalLengthMin      0
        wObjectiveFocalLengthMax      0
        wOcularFocalLength            0
        bControlSize                  3
        bmControls           0x0000000e
          Auto-Exposure Mode
[... 484 lines cut ...]
    bDevCapabilityType      4
    bReserved               0
    ContainerID             {20b69f2c-34b3-4690-a228-67bd251277e9}
  Platform Device Capability:
    bLength                28
    bDescriptorType        16
    bDevCapabilityType      5
    bReserved               0
    PlatformCapabilityUUID    {d8dd60df-4589-4cc7-9cd2-659d9e648a9f}
    CapabilityData[0]    0x00
    CapabilityData[1]    0x00
    CapabilityData[2]    0x00
    CapabilityData[3]    0x0a
    CapabilityData[4]    0x84
    CapabilityData[5]    0x00
    CapabilityData[6]    0x01
    CapabilityData[7]    0x00
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

However when I run:
```gst-launch-1.0 v4l2src device=/dev/v4l/by-path/pci-0000:00:14.0-usb-0:12:1.0-video-index0 num-buffers=10 ! image/jpeg ! multifilesink location="frame-%02d.jpg"```
IR blaster is working fine and I can see face in dumped pictures.

boltgolt, 2019-11-11:
Interesting, i think integrating your code @PetePriority could help quite a few users that have been reporting very dark IR images without the blasters firing. 

(I've moved some `lsusb` outputs into detail elements to keep the page readable)

j-m-p, 2019-11-11:
I can confirm the code by @PetePriority working to turn on the flashers on Lenovo X395.

JustAB0x, 2019-12-12:
L590 user here.

ID 04f2:b681 Chicony Electronics Co., Ltd Integrated Camera

<details>
  <summary>lsusb -d 04f2:b681 -v</summary>
                                                                                                                             
 Do 12 Dez 2019 11:39:53 CET

Bus 001 Device 004: ID 04f2:b681 Chicony Electronics Co., Ltd Integrated Camera
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.01
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x04f2 Chicony Electronics Co., Ltd
  idProduct          0xb681 
  bcdDevice           67.23
  iManufacturer           3 
  iProduct                1 
  iSerial                 2 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x04cb
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          4 
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0 
      iFunction               5 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      1 
      iInterface              5 
      VideoControl Interface Descriptor:
        bLength                13
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdUVC               1.50
        wTotalLength       0x006d
        dwClockFrequency       15.000000MHz
        bInCollection           1
        baInterfaceNr( 0)       1
      VideoControl Interface Descriptor:
        bLength                18
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0201 Camera Sensor
[... 802 lines cut ...]
        wMaxPacketSize     0x0300  1x 768 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        3
      bAlternateSetting       5
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      2 Video Streaming
      bInterfaceProtocol      1 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            5
          Transfer Type            Isochronous
          Synch Type               Asynchronous
          Usage Type               Data
        wMaxPacketSize     0x03fc  1x 1020 bytes
        bInterval               1
</details>

The patch from @PetePriority worked like a charm. IR Blaster is firing now.

morozsm, 2019-12-19:
Lenovo X1 Extreme Gen2

04f2:b67c Chicony Electronics Co., Ltd Integrated Camera
<details>
<summary>lsusb -d 04f2:b67c -v</summary>
Bus 001 Device 002: ID 04f2:b67c Chicony Electronics Co., Ltd Integrated Camera
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.01
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x04f2 Chicony Electronics Co., Ltd
  idProduct          0xb67c 
  bcdDevice           67.26
  iManufacturer           3 Chicony Electronics Co.,Ltd.
  iProduct                1 Integrated Camera
  iSerial                 2 6726
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x04cb
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          4 USB Camera
    bmAttributes         0x80
      (Bus Powered)
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass         14 Video
      bFunctionSubClass       3 Video Interface Collection
      bFunctionProtocol       0 
      iFunction               5 Integrated Camera
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass        14 Video
      bInterfaceSubClass      1 Video Control
      bInterfaceProtocol      1 
      iInterface              5 Integrated Camera
      VideoControl Interface Descriptor:
        bLength                13
        bDescriptorType        36
        bDescriptorSubtype      1 (HEADER)
        bcdUVC               1.50
        wTotalLength       0x006d
        dwClockFrequency       15.000000MHz
        bInCollection           1
        baInterfaceNr( 0)       1
      VideoControl Interface Descriptor:
        bLength                18
        bDescriptorType        36
        bDescriptorSubtype      2 (INPUT_TERMINAL)
        bTerminalID             1
        wTerminalType      0x0201 Camera Sensor
        bAssocTerminal          0
        iTerminal               0 
        wObjectiveFocalLengthMin      0
        wObjectiveFocalLengthMax      0
        wOcularFocalLength            0
[... 819 lines cut ...]
Binary Object Store Descriptor:
  bLength                 5
  bDescriptorType        15
  wTotalLength       0x0029
  bNumDeviceCaps          2
  Platform Device Capability:
    bLength                28
    bDescriptorType        16
    bDevCapabilityType      5
    bReserved               0
    PlatformCapabilityUUID    {d8dd60df-4589-4cc7-9cd2-659d9e648a9f}
    CapabilityData[0]    0x00
    CapabilityData[1]    0x00
    CapabilityData[2]    0x03
    CapabilityData[3]    0x06
    CapabilityData[4]    0x64
    CapabilityData[5]    0x03
    CapabilityData[6]    0x15
    CapabilityData[7]    0x00
  ** UNRECOGNIZED:  08 10 11 01 03 00 00 00
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
</details>
Going to try @PetePriority patch. Will report.

dsneath, 2019-12-19:
@PetePriority patch did not work on the ThinkPad TP25 (25th anniversary edition), Ubuntu 18.04, which appears to also use the Chicony camera. Building/installing patch and running 'chicony-ir-toggle on' as described on its project page results in 

> Error code: -1, errno: 2, error: No such file or directory

Is the TP25 using a different variation of the camera or am I missing a dependency? Make/build process went off without errors, only encountered at runtime.

Without blasters firing, the attached screenshot is my result. In brighter light, you can make out a silhouette, but that's it.

[image]

morozsm, 2019-12-19:
@dsneath Please review `main.c` file. There is hard coded path to IR camea `/dev/video2`. It works for ThinkPad X1 Gen2 and some other, but may be different for other laptops. Please update `main.c` accordingly with your correct IR video device.

towo, 2020-01-09:
Confirmed using the toggle code with a T490, as well.

JoaoVictorDaijo, 2020-01-29:
Sorry for the question, seems like the patch fixed the issue and makes my IR works, but how can I integrate the patch with Howdy?

JoaoVictorDaijo, 2020-01-29:
Maybe I`m going crazy but when I use the scrip on linux it break the ability of windows hello to turn on the cameras, had to restore to a backup two times in a row after using the script. Maybe I`ll just be happy with using this tool with lighting and solve the IR thing later

dsneath, 2020-01-30:
@JoaoVictorDaijo You're not crazy, I experienced that on the first reboot into Windows after running the fix as well. Mine kicked in after another reboot, though -- didn't resort to a backup.

JoaoVictorDaijo, 2020-02-06:
Well I managed to fix my issue, I went back to the forums and remembered that I have the option to say if my IR camera was working or not, so I removed Howdy installed again and cycled until it found my IR device. Now looks like its working. I also had to edit Howdy`s dark threshold to 92 for it to work fine.
(This is why you don`t do stuff half asleep)

nickedyer, 2020-03-05:
I was having the same issue and managed to get the IR blaster to turn on with @PetePriority 's solution, however I noticed something interesting. I'm running a Manjaro 19.0.1/Windows 10 dual boot on a Razer Blade 15 (Mid 2019), and I was originally able to set up howdy with no issues. The IR blaster **blinked** until it found my face and unlocked flawlessly. My issue didn't start until I booted into Windows.

I didn't have Windows Hello enabled at the time, however after messing with howdy I decided to set it up. Windows Hello worked great, and it seemed like I had no issues until booting back into Manjaro; the IR blaster stopped working all together. Like I said I managed to fix it, however I noticed that the IR blaster stopped blinking and just stayed on.

This isn't really an issue—I can change `dark_threshold` and get it working—however I'm curious as to why booting into Windows would break it. The IR blaster flashing made it easier for the IR camera to see your face without having to mess with the `dark_threshold` value. Maybe Windows messes with it somehow?

Again this is mostly food for thought, but I figured I'd share.

jmlago91, 2020-04-06:
I confirm that @PetePriority hack works on Thinkpad T495S with `/dev/video2` camera and `dark_threshold` avobe 90.

igorgad, 2020-08-01:
Hi, 
I also have a Lenovo s740, but mine came with an IMC Networks Integrated Camera. I tried to run the code made by @PetePriority to enable IR emitters but no success. I captured some USB packets on windows and I believe I found the right call, however, I could not make it work by modifying @PetePriority code.

Here is the output of `lsusb -d 13d3:56ba -v`
[lsusb_imc.txt](https://github.com/boltgolt/howdy/files/5009771/lsusb_imc.txt)

Here are the captured packets on windows
[imc_caputre.tar.gz](https://github.com/boltgolt/howdy/files/5009772/imc_caputre.tar.gz)

The packet which I believe enables the IR emitters has the following payload
```
Setup Data
    bmRequestType: 0x21
    bRequest: SET CUR (0x01)
    Control Selector: Power Line Frequency (0x05)
    Interface: 0x00
    Entity: 0x02
    wLength: 1
    Current value: 1 (0x00000001)
    bRequest: 1
    wValue: 0x0500
    wIndex: 512 (0x0200)
    wLength: 1
    Data Fragment: 01
```

Any ideas?
Thanks,
Igor.

igorgad, 2020-08-03:
Just to update. I was able to enable the IR emitters, however they do not blink. They just stay on most of the time. Does howdy use the TOF principle to create a depth image or the IR emitter is used only to illuminate the face?

Hnagzhi, 2020-08-15:
> Just to update. I was able to enable the IR emitters, however they do not blink. They just stay on most of the time. Does howdy use the TOF principle to create a depth image or the IR emitter is used only to illuminate the face?

How did you manage to turn them on?

ppriest, 2020-08-15:
Added an issue as https://github.com/PetePriority/chicony-ir-toggle/issues/11 to confirm that my Razer Blade 15" Advanced 2019 is now usable with Howdy. No changes required. However, the IR lamp isn't as bright as under Windows and I need to be much closer for it to produce usable images. You can see that the lamp is only a relatively small red dot, compared to under Windows.
I might try and do a packet capture and see if I can figure it out.

igorgad, 2020-08-15:
> > Just to update. I was able to enable the IR emitters, however they do not blink. They just stay on most of the time. Does howdy use the TOF principle to create a depth image or the IR emitter is used only to illuminate the face?
> 
> How did you manage to turn them on?

Hi Hnagzhi,
I just used [this](https://github.com/PetePriority/chicony-ir-toggle) code as-is from @PetePriority on /dev/video2. Then, just configure howdy to source video from /dev/video2 instead of /dev/video0. This way the IR emitter will turn on, but it won´t blink as windows hello.

Hnagzhi, 2020-08-15:
> Hi Hnagzhi,
> I just used [this](https://github.com/PetePriority/chicony-ir-toggle) code as-is from @PetePriority on /dev/video2. Then, just configure howdy to source video from /dev/video2 instead of [...]

Thank you for your answer! So when I run chicony-ir-toggle, it just says `Error code: -1, errno: 2, error: No such file or directory`... Meanwhile, I noticed that my device does not use exactly the same webcam model, but a slightly different one with the ID 13d3:56fb

igorgad, 2020-08-15:
Do you have /dev/video0 up to /dev/video3? Use lsusb to find out which one
is the IR camera

Hnagzhi, 2020-08-16:
> Do you have /dev/video0 up to /dev/video3? Use lsusb to find out which one is the IR camera
> […](#)

Yes. I know that video2 is the IR cam.

Tomiinek, 2020-08-29:
Great job @PetePriority ! Works even for `b6be`, but it is needed to change unit to `0x0d` and the second data byte to a higher value, i.e. `0x20`, to make it shine bright like a diamond.

bobzhu418, 2020-09-04:
> So when I run chicony-ir-toggle, it just says `Error code: -1, errno: 2, error: No such file or directory`...

When I run ` chicony-ir-toggle` on my Yoga Slim 7 AMD Ryzen 7 laptop, I get the same error message.

mdirksen, 2020-09-27:
I've also an IMC Netwroks Integrated Camera from Azurewave. lsusb : Bus 001 Device 002: ID 13d3:5406 IMC Networks Integrated Camera

I tried to write new code using documentation about IOCTL and UVCIOC_CTRL_QUERY and I noticed that the error codes for errno were not correct for the situation https://www.kernel.org/doc/html/v4.12/media/v4l-drivers/uvcvideo.html. (I also modified the buffer data to match the 9 bytes observed via Wireshark.)
Because "No such file or directory" is a generic error description. The real problem for me is "The device does not support the given control or the specified extension unit could not be found." As a result, the unit and the selector are not the right ones. So I used wireshark under a windows guest (vm) to find the unit and the selector, but it didn't work because it's a vm or I couldn't find the right ir blaster call (there's really a lot of log). 

If someone knows how to give help (have the same camera chip and use Windows), that would be nice.

EmixamPP, 2020-10-07:
https://github.com/EmixamPP/linux-enable-ir-emitter
I tried to write a tutorial to help everyone modify the PetePriority code to work with all infrared cameras (which are at least supported with the kernel driver).
I hope this will help you.

bobzhu418, 2020-10-20:
> https://github.com/EmixamPP/linux-enable-ir-emitter
> I tried to write a tutorial to help everyone modify the PetePriority code to work with all infrared cameras (which are at least supported with the kernel driver).
> I hope this will help you.

I tried your method but couldn't get it to work.

My data from VM is below:

```
Setup Data
    bmRequestType: 0x21
    bRequest: 1
    wValue: 0x1400
    wIndex: 256 (0x0100)
    wLength: 10
    Data Fragment: 00000000cf02ff040100
```

So I changed the c file:

``` 
const char* device = "/dev/video2"; //your camera path
if((fd = open(device, O_WRONLY)) < 0){
    fprintf (stderr, "Unable to open a file descriptor for %s\n", device);
    return EXIT_FAILURE;
}

//Data Fragment
__u8 data[10] = {0x00,0x00,0x00,0x00,0xcf,0x02,0xff,0x04,0x01,0x00};
//You can use my python data-seq-to-hex script for format the data fragment directly in the buffer format.
//You just have to pass the data fragment in parameter
//Don't forget to change the buffer size with the value of wLength

struct uvc_xu_control_query query = {
    .unit = 1, //2 first bits of wIndex
    .selector = 14, //2 first bits of wValue
    .query = UVC_SET_CUR,
    .size = 10, //wLength
    .data = (__u8*)&data,
};
```

When running the compiled code, I always got error:

```
Ioctl error code: -1, errno: 2
The device does not support the given control or the specified extension unit could not be found.
```

I also tried swap unit and selector, but still got the same error.

Any suggestion?

EmixamPP, 2020-10-20:
@bobzhu418 0x14 is in hex base. So you must write either 0x14 (hex) or 20 (decimal). I will update the tutorial to make it clearer.

bobzhu418, 2020-10-20:
@EmixamPP Thanks for your quick reply. I changed it but still got the same error. Is my Data Fragment part correct? 

```
    //Data Fragment
    __u8 data[10] = {0x00,0x00,0x00,0x00,0xcf,0x02,0xff,0x04,0x01,0x00};
    //You can use my python data-seq-to-hex script for format the data fragment directly in the buffer format.
    //You just have to pass the data fragment in parameter
    //Don't forget to change the buffer size with the value of wLength

    struct uvc_xu_control_query query = {
        .unit = 0x01, //2 first bits of wIndex
        .selector = 0x14, //2 first bits of wValue
        .query = UVC_SET_CUR,
        .size = 10, //wLength
        .data = (__u8*)&data,
    };
```

EmixamPP, 2020-10-20:
@bobzhu418 Have you tried another packet on wireshark ? Personally if I just open the test on windows without launching it, if the filter is applied I only have two packet which is displayed and only one of the two is the good one.

However, the steps for your camera to find the packet that activates the transmitter may be sent at another time. For example, it may be sent at the beginning of the test, so you have to launch it. Or at the moment when Windows turns on. Because with my camera Windows turns off the transmitter every time it is no longer needed but maybe not with your camera. 

I can therefore only advise you to test as many cases as possible where the transmitter could receive a packet asking it to turn on. I remain available if you need more details, either here or on the repo I created.

bobzhu418, 2020-10-21:
@EmixamPP Thank you so much for your help. I've managed to find the right setting for my IR camera. Now Howdy works great on my Manjaro setup. It did take me some time, but it was a great learning experience for me.

For anyone who may have the same laptop, my settings are below:

```
    //Data Fragment
    __u8 data[9] = {0x01,0x03,0x02,0x00,0x00,0x00,0x00,0x00,0x00};
    //You can use my python data-seq-to-hex script for format the data fragment directly in the buffer format.
    //You just have to pass the data fragment in parameter
    //Don't forget to change the buffer size with the value of wLength

    struct uvc_xu_control_query query = {
        .unit = 0x0e, //2 first bits of wIndex
        .selector = 0x06, //2 first bits of wValue
        .query = UVC_SET_CUR,
        .size = 9, //wLength
        .data = (__u8*)&data,
    };
```

renyuneyun, 2021-04-19:
I can confirm that the script by @EmixamPP works for me.
I'm on Lenovo Yoga 14s (2021), with camera `04f2:b6cb Chicony Electronics Co., Ltd Integrated Camera`.
4 video devices are shown as `/dev/videoX`, where `video0` is the normal color camera, `video2` is the IR camera.

Maybe it can be modified to read configuration file, so as to provide it as a package (e.g. AUR for archlinux)?

EmixamPP, 2021-04-20:
@renyuneyun, yes it could be really nice ! Just having a script that reads the config file and modifies the C file could help those who have difficulty with this kind of manipulation. I'll try to do that soon. 

Regarding AUR, indeed it would be cool for those who don't go on GitHub. I've never done this kind of thing, I'll have to find out or simply have someone else do it.

renyuneyun, 2021-04-22:
@EmixamPP I'm happy to help in packing it as an AUR package :) I can do the initial version and hand it to you (or maintain under my [mybuilds repository](https://github.com/renyuneyun/mybuilds) if you are reluctant in updating it yourself XD). It's a simple process once you know it.

Silverbullet069, 2023-09-20:
You guys are geniuses. Confirm Lenovo Yoga Slim 7 Pro 2020 works with @EmixamPP  script.

shanafin, 2024-02-01:
> https://github.com/EmixamPP/linux-enable-ir-emitter I tried to write a tutorial to help everyone modify the PetePriority code to work with all infrared cameras (which are at least supported with the kernel driver). I hope this will help you.

Thank you for this - your project combined with setting the dark threshold above about 95 got the IR emitter working very reliably on my Thinkpad T14 Gen 2 AMD.