boltgolt/howdy #270: Error in auth.log during authentification

Salamafet, 2019-10-24:
Hi,
When howdy try to authenticate me after a simple `sudo -i` command or in the login screen, I got this error in auth.log file:
```
Oct 24 08:44:08 Carbon14 /lib/security/howdy/pam.py[13605]: Traceback (most recent call last):
Oct 24 08:44:08 Carbon14 /lib/security/howdy/pam.py[13605]:   File "/lib/security/howdy/pam.py", line 68, in pam_sm_authenticate
Oct 24 08:44:08 Carbon14 /lib/security/howdy/pam.py[13605]:     return doAuth(pamh)
Oct 24 08:44:08 Carbon14 /lib/security/howdy/pam.py[13605]:   File "/lib/security/howdy/pam.py", line 32, in doAuth
Oct 24 08:44:08 Carbon14 /lib/security/howdy/pam.py[13605]:     sys.exit(0)
Oct 24 08:44:08 Carbon14 /lib/security/howdy/pam.py[13605]: SystemExit: 0
Oct 24 08:44:08 Carbon14 /lib/security/howdy/pam.py[13605]: Traceback (most recent call last):
Oct 24 08:44:08 Carbon14 /lib/security/howdy/pam.py[13605]:   File "/lib/security/howdy/pam.py", line 68, in pam_sm_authenticate
Oct 24 08:44:08 Carbon14 /lib/security/howdy/pam.py[13605]:     return doAuth(pamh)
Oct 24 08:44:08 Carbon14 /lib/security/howdy/pam.py[13605]:   File "/lib/security/howdy/pam.py", line 32, in doAuth
Oct 24 08:44:08 Carbon14 /lib/security/howdy/pam.py[13605]:     sys.exit(0)
Oct 24 08:44:08 Carbon14 /lib/security/howdy/pam.py[13605]: SystemExit: 0
```
I have reinstalled two times howdy because the line in `/etc/pam.d/common-auth` was not added automatically.
Finally I have add this line manually like described [here](https://github.com/boltgolt/howdy/issues/59#issuecomment-410499073).

----

I've searched for similar issues already, and my issue has not been reported yet.
I tried [this](https://github.com/boltgolt/howdy/issues/28#issuecomment-396231042) but same problem.

Linux distribution: Linux Mint 19.1 MATE
Howdy version: 2.5.1

Thanks for your help and sorry for my bad english.

Salamafet, 2019-10-30:
I have discover some things could be interesting.

Many times I use my Laptop (Lenovo ThinkPad X1 Carbon Gen 5) with Dock Station and 2 Lenovo screen with camera integrated. My Laptop is plugged with only one USB C cable and he stay close.

Howdy is configured to use one of the screen camera.

Recently I have used my laptop as a laptop. When I try the `sudo -i` command, this time before asking for password I got this message in my terminal:
```
Could not read from camera
Could not read from camera
```
I think this message are normal because howdy can't find the camera. 

So I have try to change the configuration of Howdy to use the laptop camera. And It's Works !
The problem is only with the two deported screen camera.

boltgolt, 2019-11-11:
Howdy detected that your laptops lid was closed, and aborted the auth request, as it should do. We should switch to returning PAM responses in stead of using `sys.exit()` to prevent these tracebacks in the auth.log.

closed 2020-06-22 by boltgolt