boltgolt/howdy #271: Addin a new face throws a CUDA requirement exception

rubenmulero, 2019-10-24:
Hello,

Having a laptop with an Nvidia card disabled by PRIME offloading, Howdy throws me an error when I want to add a new face:

`Traceback (most recent call last):
  File "/usr/bin/howdy", line 89, in <module>
    import cli.add
  File "/usr/lib/security/howdy/cli/add.py", line 49, in <module>
    face_encoder = dlib.face_recognition_model_v1(path + "/../dlib-data/dlib_face_recognition_resnet_model_v1.dat")
RuntimeError: Error while calling cudaGetDevice(&the_device_id) in file /home/elektro/howdy/src/dlib_clone/dlib/cuda/gpu_data.cpp:201. code: 100, reason: no CUDA-capable device is detected
`
The only way of adding a new face and being recognized by Howdy, is by activating the Nvidia card. Otherwise, the CUDA error is raised.

I'm Running Arch Linux.

Running the latest PKGBUILD downloaded from hhere

Howdy version: 2.5.1

Thanks

boltgolt, 2019-11-11:
Is the `use_cnn` config option still set to false? Otherwise, you might be using a installation of OpenCV compiled with only Nvidia support

closed 2021-04-02 by boltgolt

ccwky11, 2021-12-18:
I'm having the same issue as above, except that `use_cnn ` is disabled in config, using the `sudo howdy test` command works as expected but when it's being used for authentication or adding a new face I get this error. I have attempted to install or reinstall dlib both with and without cuda support to no avail. My device has no cuda capable GPU either, so I don't know how this suddenly began happening to me (have been using howdy for years with minimal issues).

Manjaro
Howdy version 2.6.1