boltgolt/howdy #273: Howdy error at the end of installation

BeaManCZ, 2019-10-31:
Hi, I'm new to Ubuntu and I'm trying to install howdy. At the end of installation it gives me this error:
Traceback (most recent call last):
  File "/usr/bin/pip3", line 11, in <module>
    sys.exit(main())
TypeError: 'module' object is not callable
Error in sys.excepthook:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 109, in apport_excepthook
    pr.add_proc_info(extraenv=['PYTHONPATH', 'PYTHONHOME'])
  File "/usr/lib/python3/dist-packages/apport/report.py", line 550, in add_proc_info
    self.add_proc_environ(pid, extraenv)
  File "/usr/lib/python3/dist-packages/apport/report.py", line 619, in add_proc_environ
    env = _read_file('environ', dir_fd=proc_pid_fd).replace('\n', '\\n')
  File "/usr/lib/python3/dist-packages/apport/report.py", line 73, in _read_file
    with open(path, 'rb', opener=lambda path, mode: os.open(path, mode, dir_fd=dir_fd)) as fd:
  File "/usr/lib/python3/dist-packages/apport/report.py", line 73, in <lambda>
    with open(path, 'rb', opener=lambda path, mode: os.open(path, mode, dir_fd=dir_fd)) as fd:
TypeError: argument should be integer or None, not list
Hi, I'm quite new to Linux and I'm trying to install Howdy on my laptop. But at the end of the installation it gives me this error. Could somebody help me please?

Original exception was:
Traceback (most recent call last):
  File "/usr/bin/pip3", line 11, in <module>
    sys.exit(main())
TypeError: 'module' object is not callable
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.8.7-3) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)

Thank you for any help.
Linux distribution (if applicable): Ubuntu 19.10; Howdy version: 2.5.1
Laptop: Asus Zenbook 14 UX433FN
NVIDIA GeForce MX150 2GB GDDR5; Intel Core i7 8565U Whiskey Lake

ValentinPeltier, 2019-10-31:
I got this error too

ValentinPeltier, 2019-10-31:
This solution worked for me (Ubuntu 19.10) : https://github.com/boltgolt/howdy/issues/265

boltgolt, 2019-11-11:
Indeed, duplicate of #265

closed 2019-11-11 by boltgolt