boltgolt/howdy #277: Howdy Install errors KUbuntu 19.10

mrmcreynolds, 2019-11-05:
>>> Upgrading pip to the latest version

Traceback (most recent call last):
  File "/usr/bin/pip3", line 11, in <module>
    sys.exit(main())
TypeError: 'module' object is not callable
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.8.7-3) ...
Processing triggers for libc-bin (2.30-0ubuntu2) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)

mrmcreynolds, 2019-11-06:
a PIP update this morning corrected the Python issues.  i removed Howdy via dpkg and reinstalled.  so far it is working.  will do some testing today.

AradielSabathius, 2019-11-07:
I had to remove howdy due problems at linux login and fails on some other security window login.
I will wait to the next version get solved those issues.

mrmcreynolds, 2019-11-07:
tried an HP laptop and got:

>>> Upgrading pip to the latest version

Traceback (most recent call last):
  File "/usr/bin/pip3", line 11, in <module>
    sys.exit(main())
TypeError: 'module' object is not callable
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for doc-base (0.10.9) ...
Processing 1 added doc-base file...
Processing triggers for libc-bin (2.30-0ubuntu2) ...
Processing triggers for man-db (2.8.7-3) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)

boltgolt, 2019-11-09:
Have you tried completely reinstalling pip3? This is not an issue i could fix

AradielSabathius, 2019-11-11:
That's the same issue i got on Xubuntu 19.10.
You can solve it reinstalling pip3 and python.

MarkMekhaiel, 2019-11-12:
Running Ubuntu 19.10 and am having the same issue as above.. can you explain how to reinstall pip and python3 in case I'm doing it wrong please?

I've completely purged python with `sudo apt-get purge python3`, `sudo apt-get purge python-pip`, `sudo apt-get purge python3-pip`, gone into my .local/lib files and removed everything, reinstalled from scratch with `sudo apt-get install python3`, and stil get the same error:

```
Setting up howdy (2.5.1) ...

>>> Upgrading pip to the latest version

Traceback (most recent call last):
  File "/usr/bin/pip3", line 11, in <module>
    sys.exit(main())
TypeError: 'module' object is not callable
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for libc-bin (2.30-0ubuntu2) ...
Processing triggers for man-db (2.8.7-3) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
Am I not reinstalling pip3 / python correctly? Thanks for any help and for the nice app!

MarkMekhaiel, 2019-11-12:
First of all thanks for the extremely useful app.

I'm running Ubuntu 19.10 and am having the same issue as above.. can you explain how to reinstall pip and python3 in case I'm doing it wrong please?

I've completely purged python with `sudo apt-get purge python3`, `sudo apt-get purge python-pip`, `sudo apt-get purge python3-pip`, gone into my .local/lib files and removed everything, reinstalled from scratch with `sudo apt-get install python3 python3-pip`, and stil get the same error:

```
Setting up howdy (2.5.1) ...

>>> Upgrading pip to the latest version

Traceback (most recent call last):
  File "/usr/bin/pip3", line 11, in <module>
    sys.exit(main())
TypeError: 'module' object is not callable
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for libc-bin (2.30-0ubuntu2) ...
Processing triggers for man-db (2.8.7-3) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Am I not reinstalling pip3 / python correctly? Thanks in advance

AradielSabathius, 2019-11-13:
I suggest you to update also your actual python version

przemek-pokrywka, 2019-11-29:
The reason of the TypeError is that `/usr/bin/pip3` in Ubuntu is not up to date with PIP changes. Until that's fixed, you can use the workaround from https://github.com/boltgolt/howdy/issues/265#issuecomment-545704230

closed 2026-11-16 by boltgolt