boltgolt/howdy #279: Install errors - Ubuntu 19.10 after reinstalling Python + PIP

MarkMekhaiel, 2019-11-12:
First of all thanks for the extremely useful app.

I'm running Ubuntu 19.10 and am having the same issue as above.. can you explain how to reinstall pip and python3 in case I'm doing it wrong please?

I've completely purged python with `sudo apt-get purge python3`, `sudo apt-get purge python-pip`, `sudo apt-get purge python3-pip`, gone into my .local/lib files and removed everything, reinstalled from scratch with `sudo apt-get install python3 python3-pip`, and stil get the same error:

```
Setting up howdy (2.5.1) ...

>>> Upgrading pip to the latest version

Traceback (most recent call last):
  File "/usr/bin/pip3", line 11, in <module>
    sys.exit(main())
TypeError: 'module' object is not callable
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for libc-bin (2.30-0ubuntu2) ...
Processing triggers for man-db (2.8.7-3) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Am I not reinstalling pip3 / python correctly? Thanks in advance

boltgolt, 2019-11-12:
I've seen your comment, no need to make a new issue

boltgolt, 2019-11-12:
I'll reply to your original comment when i figure out how else to reinstall pip

closed 2019-11-12 by boltgolt

MarkMekhaiel, 2019-11-12:
My bad - will reinstate my earlier comment