boltgolt/howdy #281: Camera path is not configured correctly,

Sobuj53, 2019-11-14:
Hello,
I've installed Howdy on my Manjaro today. It was bit unconventional way as I couldn't used the AUR version (it was outdated). Anyway I used PKGBUILD from here [https://raw.githubusercontent.com/boltgolt/howdy/22891a1a1b5ffc8a6311199141a85fd3aa00df0d/archlinux/howdy/PKGBUILD](url). After adding missing dependency the installation worked, but now I can't use it. When I run `sudo howdy test` it gives me following error,

> [ WARN:0] global /build/opencv/src/opencv-4.1.2/modules/videoio/src/cap_gstreamer.cpp (711) open OpenCV | GStreamer warning: Error opening bin: no element "none"
[ WARN:0] global /build/opencv/src/opencv-4.1.2/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[ERROR:0] global /build/opencv/src/opencv-4.1.2/modules/videoio/src/cap.cpp (116) open VIDEOIO(CV_IMAGES): raised OpenCV exception:

> OpenCV(4.1.2) /build/opencv/src/opencv-4.1.2/modules/videoio/src/cap_images.cpp:253: error: (-5:Bad argument) CAP_IMAGES: can't find starting number (in the name of file): none in function 'icvExtractPattern'

> Opening a window with a test feed

> Press ctrl+C in this terminal to quit

> Click on the image to enable or disable slow mode

> QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'

> Unknown camera, please check your 'device_path' config value.

> Traceback (most recent call last):
  File "/usr/bin/howdy", line 101, in <module>
    import cli.test
  File "/usr/lib/security/howdy/cli/test.py", line 117, in <module>
    frame = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.1.2) /build/opencv/src/opencv-4.1.2/modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'

----

I've searched for similar issues already, and my issue has not been reported yet.

My media UID is: /dev/vedio0
My WebCam is: HP HD WebCam

Linux distribution (if applicable): Manjaro KDE based on Arch Linux

Howdy version: 2.5.1

I'd really appreciate you help on this. Thank you.

boltgolt, 2019-11-14:
Did you edit the 'device_path' config value?

Sobuj53, 2019-11-14:
No, how do I do that?

boltgolt, 2019-11-14:
Try running `sudo howdy config`

Sobuj53, 2019-11-14:
Hi,
Thank you for your replay. 
[config.txt](https://github.com/boltgolt/howdy/files/3845967/config.txt)
I've attached my howdy Config file. the device is set to none `device_path = none`, how do i identify my device path?. I'd really appreciate your feedback on this.

boltgolt, 2019-11-14:
Checkout [this part](https://wiki.archlinux.org/index.php/Howdy#Add_correct_IR_sensor) of the arch wiki article

Sobuj53, 2019-11-14:
Ok I just set my `device_path = /dev/vedio0` and now it's working. Thanks. But really wish there is a easier way to install it in Manjaro. Thank you for everything

boltgolt, 2019-11-14:
I'm trying to make it easier, see #262. Good to hear that it's fixed though!

closed 2019-11-14 by boltgolt

ajay139, 2020-06-29:
Yes it work for me too after changing the device path to /dev/video0
But I would like suggest to install v4l-utils before installing howdy so that we could identify the camera device path by performing v4l2-ctl --list-devices

GarvTambi, 2020-07-07:
When I run this command after changing the config_path
I got this error:
Adding face model for the user garv
Enter a label for this new model [Initial model] (max 24 characters): Facelock
Camera path is not configured correctly, please edit the 'device_path' config value.
Exception ignored in: <bound method VideoCapture.__del__ of <recorders.video_capture.VideoCapture object at 0x7f36c737d4a8>>
Traceback (most recent call last):
  File "/lib/security/howdy/recorders/video_capture.py", line 55, in __del__
    self.internal.release()
AttributeError: 'VideoCapture' object has no attribute 'internal'

Please Help me!!

ajaykrishnansm, 2020-07-08:
> When I run this command after changing the config_path
> I got this error:
> Adding face model for the user garv
> Enter a label for this new model [Initial model] (max 24 characters): Facelock
> Camera [...]

Have you correctly spelled the device path
Make sure you spell correct device path to your preffered video device
You can check the correct device path by installing v4l-utils
And perform
v4l2-ctl -- list-devices

GarvTambi, 2020-07-08:
> Have you correctly spelled the device path?
> Make sure you spell correct device path to your preffered video device
> You can check the correct device path by installing v4l-utils
> And perform
> v4l2-ctl [...]

How I install v4l-utils

ajaykrishnansm, 2020-07-08:
> How I install v4l-utils

sudo apt-get install v4l-utils

GarvTambi, 2020-07-08:
(base) garv@garv-Inspiron-5567:~$ sudo apt install v4l-utils
Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 17, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'
Unknown error: 1
[sudo] password for garv: 
Sorry, try again.
Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 17, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'
Unknown error: 1

Please help me!!

GarvTambi, 2020-07-08:
(base) garv@garv-Inspiron-5567:~$ sudo apt-get install v4l-utils
Reading package lists... Done
Building dependency tree       
Reading state information... Done
v4l-utils is already the newest version (1.14.2-1).
The following packages were automatically installed and are no longer required:
  libegl1-mesa libfwup1
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 39 not upgraded.

Done!! Thanks a lot
Now what is the next step

ajaykrishnansm, 2020-07-08:
> (base) garv@garv-Inspiron-5567:~$ sudo apt-get install v4l-utils
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> v4l-utils is already the newest version [...]

v4l2-ctl -- list-devices

GarvTambi, 2020-07-08:
(base) garv@garv-Inspiron-5567:~$ v4l2-ctl -- list-devices
unknown arguments: list-devices 

General/Common options:
  --all              display all information available
  -C, --get-ctrl=<ctrl>[,<ctrl>...]
                     get the value of the controls [VIDIOC_G_EXT_CTRLS]
  -c, --set-ctrl=<ctrl>=<val>[,<ctrl>=<val>...]
                     set the value of the controls [VIDIOC_S_EXT_CTRLS]
  -D, --info         show driver info [VIDIOC_QUERYCAP]
  -d, --device=<dev> use device <dev> instead of /dev/video0
                     if <dev> starts with a digit, then /dev/video<dev> is used
  -e, --out-device=<dev> use device <dev> for output streams instead of the
                     default device as set with --device
                     if <dev> starts with a digit, then /dev/video<dev> is used
  -h, --help         display this help message
  --help-all         all options
  --help-io          input/output options
  --help-misc        miscellaneous options
  --help-overlay     overlay format options
  --help-sdr         SDR format options
  --help-selection   crop/selection options
  --help-stds        standards and other video timings options
  --help-streaming   streaming options
  --help-tuner       tuner/modulator options
  --help-vbi         VBI format options
  --help-vidcap      video capture format options
  --help-vidout      vidout output format options
  --help-edid        edid handling options
  -k, --concise      be more concise if possible.
  -l, --list-ctrls   display all controls and their values [VIDIOC_QUERYCTRL]
  -L, --list-ctrls-menus
		     display all controls and their menus [VIDIOC_QUERYMENU]
  -r, --subset=<ctrl>[,<offset>,<size>]+
                     the subset of the N-dimensional array to get/set for control <ctrl>,
                     for every dimension an (<offset>, <size>) tuple is given.
  -w, --wrapper      use the libv4l2 wrapper library.
  --list-devices     list all v4l devices
  --log-status       log the board status in the kernel log [VIDIOC_LOG_STATUS]
  --get-priority     query the current access priority [VIDIOC_G_PRIORITY]
  --set-priority=<prio>
                     set the new access priority [VIDIOC_S_PRIORITY]
                     <prio> is 1 (background), 2 (interactive) or 3 (record)
  --silent           only set the result code, do not print any messages
  --sleep=<secs>     sleep <secs>, call QUERYCAP and close the file handle
  --verbose          turn on verbose ioctl status reporting

I get this...Now what is the next step?

GarvTambi, 2020-07-08:
> I get this...Now what is the next step?

ajaykrishnansm, 2020-07-08:
> > I get this...Now what is the next step?

v4l2-ctl --list-devices
Have you typed it correctly
If so,then it should be providing you with the video device path

ajaykrishnansm, 2020-07-08:
> > > I get this...Now what is the next step?
> 
> v4l2-ctl --list-devices
> Have you typed it correctly
> If so,then it should be providing you with the video device path

[image]

GarvTambi, 2020-07-08:
It provide me this

default device as set with --device
if starts with a digit, then /dev/video is used

> (base) garv@garv-Inspiron-5567:~$ v4l2-ctl -- list-devices
> unknown arguments: list-devices
> 
> General/Common options:
> --all display all information available
> -C, --get-ctrl=[,...]
> get the value of the controls [VIDIOC_G_EXT_CTRLS]
> -c, --set-ctrl==[,=...]
> set the value of the controls [VIDIOC_S_EXT_CTRLS]
> -D, --info show driver info [VIDIOC_QUERYCAP]
> -d, --device= use device instead of /dev/video0
> if starts with a digit, then /dev/video is used
> -e, --out-device= use device for output streams instead of the
> default device as set with --device
> if starts with a digit, then /dev/video is used
> -h, --help display this help message
> --help-all all options
> --help-io input/output options
> --help-misc miscellaneous options
> --help-overlay overlay format options
> --help-sdr SDR format options
> --help-selection crop/selection options
> --help-stds standards and other video timings options
> --help-streaming streaming options
> --help-tuner tuner/modulator options
> --help-vbi VBI format options
> --help-vidcap video capture format options
> --help-vidout vidout output format options
> --help-edid edid handling options
> -k, --concise be more concise if possible.
> -l, --list-ctrls display all controls and their values [VIDIOC_QUERYCTRL]
> -L, --list-ctrls-menus
> display all controls and their menus [VIDIOC_QUERYMENU]
> -r, --subset=[,,]+
> the subset of the N-dimensional array to get/set for control ,
> for every dimension an (, ) tuple is given.
> -w, --wrapper use the libv4l2 wrapper library.
> --list-devices list all v4l devices
> --log-status log the board status in the kernel log [VIDIOC_LOG_STATUS]
> --get-priority query the current access priority [VIDIOC_G_PRIORITY]
> --set-priority=
> set the new access priority [VIDIOC_S_PRIORITY]
> is 1 (background), 2 (interactive) or 3 (record)
> --silent only set the result code, do not print any messages
> --sleep= sleep , call QUERYCAP and close the file handle
> --verbose turn on verbose ioctl status reporting
> 

It provide me this:
-d, --device= use device instead of /dev/video0
if starts with a digit, then /dev/video is used

ajaykrishnansm, 2020-07-08:
> > > > I get this...Now what is the next step?
> > 
> > 
> > v4l2-ctl --list-devices
> > Have you typed it correctly
> > If so,then it should be providing you with the video device path
> 
> [image]

Then perform 
sudo howdy config

This is my selected device path:
[image]

Then save it.
Hope this will work for you.

GarvTambi, 2020-07-08:
Yess, I get this

(base) garv@garv-Inspiron-5567:~$ v4l2-ctl --list-devices
Integrated_Webcam_HD: Integrate (usb-0000:00:14.0-5):
	/dev/video0
	/dev/video1

Now what is the next step??

GarvTambi, 2020-07-08:
when I run sudo howdy cofig...I get this error:

(base) garv@garv-Inspiron-5567:~$ sudo howdy config
Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 17, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'
Unknown error: 1
[sudo] password for garv:

ajaykrishnansm, 2020-07-08:
> Yess, I get this
> 
> (base) garv@garv-Inspiron-5567:~$ v4l2-ctl --list-devices
> Integrated_Webcam_HD: Integrate (usb-0000:00:14.0-5):
> /dev/video0
> /dev/video1
> 
> Now what is the next step??

Perform 
sudo howdy config

Find line start with
device_path 
May be in your case
This will be
device_path = none
.
change to 

device_path = /dev/video0

ajaykrishnansm, 2020-07-08:
> when I run sudo howdy cofig...I get this error:
> 
> (base) garv@garv-Inspiron-5567:~$ sudo howdy config
> Traceback (most recent call last):
> File "/lib/security/howdy/compare.py", line 17, in
> import cv2
> ModuleNotFoundError: No module named 'cv2'
> Unknown error: 1
> [sudo] password for garv:

Its prompting for your password
Give it and check

ajaykrishnansm, 2020-07-08:
> Its prompting for your password
> Give it and check

sudo howdy config

Give your password

Then find the line start with 
device_path
Change the value 
device_path = /dev/video0
Save it then check if it is working or not

GarvTambi, 2020-07-08:
Yess Sir, I save it!! and it save successfully.

Now what is the next step?? 

> sudo howdy config
>
> Give your password
>
> Then find the line start with
> device_path
> Change the value
> device_path = /dev/video0
> Save it then check if it is working or not

ajaykrishnansm, 2020-07-08:
> Yess Sir, I save it!! and it save successfully.
>
> Now what is the next step??

Try
sudo howdy add

GarvTambi, 2020-07-08:
Sir Done!!

Adding face model for the user garv
Enter a label for this new model [Initial model] (max 24 characters): FaceLock

Please look straight into the camera
Scan complete

Added a new model to garv

Now What is the next step??

GarvTambi, 2020-07-08:
> Now What is the next step??

Now What is the next step??

ajaykrishnansm, 2020-07-08:
> Sir Done!!
> 
> Adding face model for the user garv
> Enter a label for this new model [Initial model] (max 24 characters): FaceLock
> 
> Please look straight into the camera
> Scan complete
> 
> Added a new model to garv
> 
> Now What is the next step??

That's it 
you have added your face
And try reboot your and check whether your front camera helps you to unlock your system

GarvTambi, 2020-07-08:
Sir Unknown error is shown while testing

[image]

ajaykrishnansm, 2020-07-08:
> Sir Unknown error is shown while testing
> 
> [image]

Don't have an answer for this😁 , because  I didn't face such issues
Have you correctly followed the installation instruction?
Try updating the system
Install 
pip pip3 
Or reinstalling howdy
https://youtu.be/3AG_RST6o1Y

GarvTambi, 2020-07-08:
> > Sir Unknown error is shown while testing
> > [image]
> 
> Don't have an answer for this , because I didn't face such issues
> Have you correctly followed the installation instruction?
> Try updating the system
> Install
> pip pip3
> Or reinstalling howdy
> https://youtu.be/3AG_RST6o1Y

Sir, May I run this command again 'sudo howdy add'

ajaykrishnansm, 2020-07-08:
> Sir, May I run this command again 'sudo howdy add'

Yeah,you can try it

GarvTambi, 2020-07-08:
> sudo howdy add

I think this will happen because my camera quality is like this:

[image]

GarvTambi, 2020-07-08:
> > sudo howdy add
> 
> I think this will happen because my camera quality is like this:
> 
> [image]

What you think about this??

ajaykrishnansm, 2020-07-08:
> > sudo howdy add
> 
> I think this will happen because my camera quality is like this:
> 
> [image]

Not at all
It because something went wrong during the installation.

GarvTambi, 2020-07-08:
So How I reinstall all these stuff??

ajaykrishnansm, 2020-07-08:
> So How I reinstall all these stuff??

sudo apt-get remove howdy
sudo apt-get purge howdy
then reinstall again by correctly following the instructions in
https://github.com/boltgolt/howdy

ajaykrishnansm, 2020-07-08:
> > So How I reinstall all these stuff??
> 
> sudo apt-get remove howdy
> sudo apt-get purge howdy
> then reinstall again by correctly following the instructions in
> https://github.com/boltgolt/howdy

Hope next time the problem will not occur

GarvTambi, 2020-07-08:
> > > So How I reinstall all these stuff??
> > 
> > 
> > sudo apt-get remove howdy
> > sudo apt-get purge howdy
> > then reinstall again by correctly following the instructions in
> > https://github.com/boltgolt/howdy
> 
> Hope next time the problem will not occur

Hopefully!!

GarvTambi, 2020-07-11:
Getting this erro while running 'sudo apt install howdy' on terminal...

Traceback (most recent call last):
  File "/var/lib/dpkg/info/howdy.postinst", line 159, in <module>
    tf.extract(item, "/tmp")
  File "/usr/lib/python3.6/tarfile.py", line 2052, in extract
    numeric_owner=numeric_owner)
  File "/usr/lib/python3.6/tarfile.py", line 2122, in _extract_member
    self.makefile(tarinfo, targetpath)
  File "/usr/lib/python3.6/tarfile.py", line 2171, in makefile
    copyfileobj(source, target, tarinfo.size, ReadError, bufsize)
  File "/usr/lib/python3.6/tarfile.py", line 255, in copyfileobj
    buf = src.read(remainder)
  File "/usr/lib/python3.6/gzip.py", line 276, in read
    return self._buffer.read(size)
  File "/usr/lib/python3.6/_compression.py", line 68, in readinto
    data = self.read(len(byte_view))
  File "/usr/lib/python3.6/gzip.py", line 482, in read
    raise EOFError("Compressed file ended before the "
EOFError: Compressed file ended before the end-of-stream marker was reached
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)

ajaykrishnansm, 2020-07-11:
> Getting this erro while running 'sudo apt install howdy' on terminal...
> 
> Traceback (most recent call last):
> File "/var/lib/dpkg/info/howdy.postinst", line 159, in
> tf.extract(item, "/tmp")
> File "/usr/lib/python3.6/tarfile.py", line 2052, in extract
> numeric_owner=numeric_owner)
> File "/usr/lib/python3.6/tarfile.py", line 2122, in _extract_member
> self.makefile(tarinfo, targetpath)
> File "/usr/lib/python3.6/tarfile.py", line 2171, in makefile
> copyfileobj(source, target, tarinfo.size, ReadError, bufsize)
> File "/usr/lib/python3.6/tarfile.py", line 255, in copyfileobj
> buf = src.read(remainder)
> File "/usr/lib/python3.6/gzip.py", line 276, in read
> return self._buffer.read(size)
> File "/usr/lib/python3.6/_compression.py", line 68, in readinto
> data = self.read(len(byte_view))
> File "/usr/lib/python3.6/gzip.py", line 482, in read
> raise EOFError("Compressed file ended before the "
> EOFError: Compressed file ended before the end-of-stream marker was reached
> dpkg: error processing package howdy (--configure):
> installed howdy package post-installation script subprocess returned error exit status 1
> Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
> Errors were encountered while processing:
> howdy
> E: Sub-process /usr/bin/dpkg returned an error code (1)

try
sudo apt autoremove
sudo apt-get howdy

May be it will work

GarvTambi, 2020-07-11:
In (f,b,s ) which one I choose

ajaykrishnansm, 2020-07-11:
> In (f,b,s ) which one I choose

S

GarvTambi, 2020-07-13:
Sometimes face detect successfully and sometimes it shows Unknown Error:1 Why?
Please help!

[image]

mefy-garv, 2020-08-26:
Is it possible to add multiple faces in a same system.

phamthainb, 2020-11-07:
> Ok I just set my `device_path = /dev/vedio0` and now it's working. Thanks. But really wish there is a easier way to install it in Manjaro. Thank you for everything

`device_path = /dev/video0` work for me.

RedCoder09, 2020-12-24:
Hello, when I run sudo howdy add I get this:
No face model known
[sudo] password for redcoder09: 
Sorry, try again.
No face model known
[sudo] password for redcoder09: 
Adding face model for the user redcoder09
Enter a label for this new model [Initial model] (max 24 characters): me

Please look straight into the camera
Failed to read camera specified in your 'device_path', aborting

Any idea why?

ALENTL, 2021-04-25:
> Checkout [this part](https://wiki.archlinux.org/index.php/Howdy#Add_correct_IR_sensor) of the arch wiki article
SO what about in ubuntu based distros