boltgolt/howdy #283: Latest python 3.8 breaks howdy

O-Zone-Guy, 2019-11-15:
_Please describe the issue in as much detail as possible, including any errors and traces_

Latest update to python 3.8 breaks howdy.

When using howdy, python error messages are outputted about `dlib`, I believe, not being available.

I tried uninstalling it then reinstalling, but reinstalling failed with `Py_DEPRECATED(3.7) PyAPI_FUNC(int) PyThread_set_key_value(int key` all over the output.

It was working before the update and now it doesn't.

Final error when reinstalling:
```
The --yes options to dlib's setup.py don't do anything since all these options 
are on by default.  So --yes has been removed.  Do not give it to setup.py.
==> ERROR: A failure occurred in package().
    Aborting...
```

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Arch linux

Howdy version: Cannot install...

boltgolt, 2019-11-15:
The PKGBUILD is really outdated at the moment, check out #262

closed 2019-11-15 by boltgolt

kageurufu, 2019-11-15:
@O-Zone-Guy The AUR has now been updated, let us know if you still have problems with it

sogeil, 2022-01-08:
When trying to install Howdy on bodhi 6.0 (Ubuntu 20.04), the above message keeps coming back some time after the query regarding (fast / balanced / secure):

linux howdy Py_DEPRECATED (3.7) PyAPI_FUNC (int) ...

The PC is almost completely frozen, nothing worked anymore, I had to switch it off.
I followed the instructions:
https://itsfoss.com/face-unlock-ubuntu/

What I have done wrong?