boltgolt/howdy #285: Error with Howdy test

Raymo111, 2019-11-16:
_Please describe the issue in as much detail as possible, including any errors and traces_
Error on 'howdy test':
```
 howdy test
[ WARN:0] global /build/opencv/src/opencv-4.1.2/modules/videoio/src/cap_gstreamer.cpp (1756) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0] global /build/opencv/src/opencv-4.1.2/modules/videoio/src/cap_gstreamer.cpp (886) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /build/opencv/src/opencv-4.1.2/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
Traceback (most recent call last):
  File "/usr/bin/howdy", line 101, in <module>
    import cli.test
  File "/usr/lib/security/howdy/cli/test.py", line 206, in <module>
    time.sleep(.5 - frame_time)
ValueError: sleep length must be non-negative
```
Also, my cpu shoots to 100%, and it does not work when I log in (i3lock, sddm, etc.)
----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):
Arch
Howdy version:
2.5.1-2

boltgolt, 2019-11-19:
Impressive, that means you're using more than 500ms to process a single frame. Could you try to check which call between line 111 and 202 in test.py takes so long?

Raymo111, 2019-11-19:
Err... how would I go about doing that?

boltgolt, 2019-11-20:
I'm not sure how good you are with python, but you can insert the following line and move it down a line every time:

```
print(time.time() - frame_tm)
```

This will print the execution time since line 111. Make sure you use the right indentation

Raymo111, 2019-11-20:
Ooh thanks! I'll try that.

closed 2024-04-15 by musikid