boltgolt/howdy #286: Howdy CLI not working on Ubuntu 18.04

NtsikaMngoma, 2019-11-18:
_I am looking to setup howdy on Ubuntu Bionic Beaver but the command is not recognized after a successful installation. I am quite confused what the issue might be. It give us back:_

howdy: command not found and when I run whereis howdy, it shows the .gz path

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu 18.04

Howdy version: 2.5.1

M0Rph3U56031769, 2019-11-21:
how did you install it? I got this error:
Attempting installation of missing data files
Downloading 3 required data files...
Unpacking...
bzip2: Can't open input file *.bz2: No such file or directory.
Error while running last command

boltgolt, 2019-11-27:
Did you install it from the PPA?

---

@M0Rph3U56031769 That's a different issue, retry the installation and open a new issue if it persists

thhart, 2019-12-19:
This is what I get in an Ubuntu 19.10. It is from the ppa. Dlib compile goes ok. Camera is detected but finally fails to create the execute. Looks like a problem in pip3:

```
Traceback (most recent call last):
  File "/usr/bin/pip3", line 11, in <module>
    sys.exit(main())
TypeError: 'module' object is not callable
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.8.7-3) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

boltgolt, 2019-12-20:
@thhart That's also a completely different problem, see #227 and please use the search function if possible

Closing as it's attracting unrelated issues and the OP has not responded

closed 2019-12-20 by boltgolt

yimisunrise, 2020-08-03:
@NtsikaMngoma 
I have the same problem ，I've tried to reinstall it many times and still do 。
Have you finally solved it ？

matifali, 2020-09-02:
This worked for me
```
sudo apt remove howdy
sudo pip3 install --upgrade pip
sudo dpkg --install howdy_2.6.0.deb
```

gaoshuai, 2020-10-23:
great