boltgolt/howdy #287: How many face models do you have?

papermache, 2019-11-18:
I have 13 and would rate my Howdy as slightly unreliable. For those experiencing good performance are 13 more or less models than what you have?

boltgolt, 2019-11-19:
I believe from my testing that anything above 6 models is hurting recognition more than it helps. Howdy can check less frames because it has to go through every model for every frame first, and the improvement is recognition diminishes the more you add.

I'd suggest upping the `certainty` config value instead

papermache, 2019-11-19:
Six models, certainty at 5, it's not a matter of security rather urgency if I really need access. See `[https://github.com/boltgolt/howdy/issues/9#issuecomment-555161840]

closed 2019-11-19 by papermache