boltgolt/howdy #290: Error while run "sudo howdy config" with no nano editor in Fedora 31

DnsZhou, 2019-11-21:
When running `sudo howdy config` command, it comes up with the error listed below:
```
Opening config.ini in the default editor
Traceback (most recent call last):
  File "/usr/bin/howdy", line 93, in <module>
    import cli.config
  File "/usr/lib64/security/howdy/cli/config.py", line 20, in <module>
    subprocess.call([editor, os.path.dirname(os.path.realpath(__file__)) + "/../config.ini"])
  File "/usr/lib64/python3.7/subprocess.py", line 339, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib64/python3.7/subprocess.py", line 800, in __init__
    restore_signals, start_new_session)
  File "/usr/lib64/python3.7/subprocess.py", line 1551, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/bin/nano': '/bin/nano'
```
In `/usr/lib64/security/howdy/cli/config.py`:

```
#Open the config file in an editor

# Import required modules
import os
import subprocess

# Let the user know what we're doing
print("Opening config.ini in the default editor")

# Default to the nano editor
editor = "/bin/nano"

# Use the user preferred editor if available
if os.path.isfile("/etc/alternatives/editor"):
        editor = "/etc/alternatives/editor"
elif "EDITOR" in os.environ:
        editor = os.environ["EDITOR"]

# Open the editor as a subprocess and fork it
subprocess.call([editor, os.path.dirname(os.path.realpath(__file__)) + "/../config.ini"])

```
Seems it did not handle the case that there is no nano editor in the system.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):

**Fedora 31** 

Howdy version:
**2.5.1**

boltgolt, 2019-11-21:
Howdy already tries to get a preferred editor from both `/etc/alternatives` and the environment, and only if both fail will it fallback to nano. I don't think there's much more to do without writing way more code than required for 99% of users

closed 2019-11-27 by boltgolt

jpcaparas, 2024-06-18:
For Fedora 40 users, of which a fresh install doesn't have an `editor` entry.

```
sudo update-alternatives --install /usr/bin/editor editor $(which vim) 100
```