boltgolt/howdy #291: XFCE4 Lock screen

pyro57, 2019-11-27:
_Please describe the issue in as much detail as possible, including any errors and traces_
I'm running XFCE4 on Arch using GDM as the display manger, it works perfectly there, and when I sign into gnome it works flawlessly.  However in an XFCE session the lock screen doesn't respect the pam settings.  Every other kind of authentication does (ex: sudo, gsudo, ect ect..) does just not the lock screen.  I've added the line for howdey in all the lightdm pam files as well as the xfce4 screensaver pam file, and I'm not sure where else I can look for settings.

Thank you

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Arch Linux

Howdy version: Unsure, might be a version or two behind.

lcoronelp, 2020-06-20:
I have the same issue in xfce4 over xubuntu 20.04.

The /var/log/auth.log show that:

```
Jun 20 18:03:30 phoenix /lib/security/howdy/pam.py[5576]: Traceback (most recent call last):
Jun 20 18:03:30 phoenix /lib/security/howdy/pam.py[5576]:   File "/lib/security/howdy/pam.py", line 68, in pam_sm_authenticate
Jun 20 18:03:30 phoenix /lib/security/howdy/pam.py[5576]:     return doAuth(pamh)
Jun 20 18:03:30 phoenix /lib/security/howdy/pam.py[5576]:   File "/lib/security/howdy/pam.py", line 62, in doAuth
Jun 20 18:03:30 phoenix /lib/security/howdy/pam.py[5576]:     pamh.conversation(pamh.Message(pamh.PAM_ERROR_MSG, "Unknown error: " + str(status)))
Jun 20 18:03:30 phoenix /lib/security/howdy/pam.py[5576]: PamException: Error de conversación
Jun 20 18:03:30 phoenix xfce4-screensaver-dialog[5576]: pam_unix(xfce4-screensaver:auth): Couldn't open /etc/securetty: No existe el archivo o el directorio
Jun 20 18:03:30 phoenix xfce4-screensaver-dialog[5619]: pam_unix(xfce4-screensaver:auth): Couldn't open /etc/securetty: No existe el archivo o el directorio
Jun 20 18:03:34 phoenix xfce4-screensaver-dialog[5652]: pam_unix(xfce4-screensaver:auth): Couldn't open /etc/securetty: No existe el archivo o el directorio
```

amdist, 2020-08-30:
Seeing the same on Ubuntu Budgie 20.04

Aug 30 08:30:14 amdi-bud /lib/security/howdy/pam.py[15473]: Traceback (most recent call last):
Aug 30 08:30:14 amdi-bud /lib/security/howdy/pam.py[15473]:   File "/lib/security/howdy/pam.py", line 84, in pam_sm_authenticate
Aug 30 08:30:14 amdi-bud /lib/security/howdy/pam.py[15473]:     return doAuth(pamh)
Aug 30 08:30:14 amdi-bud /lib/security/howdy/pam.py[15473]:   File "/lib/security/howdy/pam.py", line 77, in doAuth
Aug 30 08:30:14 amdi-bud /lib/security/howdy/pam.py[15473]:     pamh.conversation(pamh.Message(pamh.PAM_ERROR_MSG, "Unknown error: " + str(status)))
Aug 30 08:30:14 amdi-bud /lib/security/howdy/pam.py[15473]: PamException: Konversationsfejl
Aug 30 08:30:14 amdi-bud gnome-screensaver-dialog[15473]: pam_unix(gnome-screensaver:auth): Couldn't open /etc/securetty: Ingen sådan fil eller filkatalog
Aug 30 08:30:14 amdi-bud gnome-screensaver-dialog[15485]: pam_unix(gnome-screensaver:auth): Couldn't open /etc/securetty: Ingen sådan fil eller filkatalog
Aug 30 08:30:18 amdi-bud gnome-screensaver-dialog[15488]: pam_unix(gnome-screensaver:auth): Couldn't open /etc/securetty: Ingen sådan fil eller filkatalog
A

patrakov, 2021-09-10:
Same on cinnamon-screensaver. Commenting out all "pamh.conversation" lines fixes this.

chenxiex, 2022-08-05:
Same on arch+lightdm+xfce4. Howdy works on lightdm, but not on xflock.

closed 2026-11-16 by boltgolt