boltgolt/howdy #296: Howdy for nixos

Th0rgal, 2019-12-18:
Hello there,

I'm using nixos on my laptop, it's a really good system and I love it but howdy isn't available for it atm. Could you add it to nixpkgs?

Thank you very much,
Thomas

boltgolt, 2019-12-18:
I don't have a nixos system to test packaging on, but it looks like you could change the debian package structure a little and make it installable

closed 2020-01-10 by boltgolt

LEXUGE, 2020-04-03:
I have got one packaged with others and it works out well
In this [folder](https://github.com/LEXUGE/nixos/tree/master/packages), you will need `howdy.nix`, `howdy.patch`, `pam_python.nix` and `libpam-wrapper.nix`

Th0rgal, 2020-04-03:
Great, I'll try it asap! Thank you.

Th0rgal, 2020-04-21:
I've been very busy recently and I haven't had time to test, unfortunately your links don't work anymore.

LEXUGE, 2020-04-21:
> I've been very busy recently and I haven't had time to test, unfortunately your links don't work anymore.

Yes, because I restructured it and it is even easier to test. See home page and [here](https://github.com/LEXUGE/nixos/tree/master/plugins/devices/howdy)

Th0rgal, 2020-04-21:
This looks great but I need icebox to use it and unfortunately icebox is only unstable branch (I'm on 20.03). I tried to install it anyway but I got an error:
```
building Nix...
building the system configuration...
error:
Failed assertions:
- Neither the root account nor any wheel user has a password or SSH authorized key.
You must set one to prevent being locked out of your system.
(use '--show-trace' to show detailed location information)
```

But my root account does have a password.

LEXUGE, 2020-04-21:
> This looks great but I need icebox to use it and unfortunately icebox is only unstable branch (I'm on 20.03). I tried to install it anyway but I got an error:
>
> ```
> building Nix...
> building the [...]

This has nothing to do with unstable, I have not investigated that, you could see this [tree](https://github.com/LEXUGE/nixos/tree/88b3a887167c88b5407e900966fdef9e6f38c575) to overlay howdy and other packages.

SpiderUnderUrBed, 2025-02-25:
PR being worked on here:
https://github.com/NixOS/nixpkgs/pull/216245