boltgolt/howdy #297: Can't use any function

dannySOLO, 2019-12-23:
[image]
_Please describe the issue in as much detail as possible, including any errors and traces_
I don't know exactly what are these problems.
* sudo howdy add: no face detected, aborting
* sudo howdy test: a grey screen
* I'm using Lenovo Ideapad 500 Isk.
Please help! :((

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):

Howdy version:

boltgolt, 2020-01-07:
What's the full output in the terminal?

dannySOLO, 2020-01-10:
The screenshot I made was all of the output when I ran "test".
The camera lights were on, but no function worked.

boltgolt, 2020-01-10:
Okay, what was the full output in your terminal (text cut off in your screenshot)

dannySOLO, 2020-01-10:
[image]
This is when I run 'howdy test'.

dannySOLO, 2020-01-11:
[image]

boltgolt, 2021-04-02:
Sorry this issue must have slipped through my email at the time

closed 2021-04-02 by boltgolt