boltgolt/howdy #300: GLib-GObject-CRITICAL

Sobuj53, 2019-12-31:
_Please describe the issue in as much detail as possible, including any errors and traces_
Hello I'm running Howdy 2.5.1-2 in Manjaro KDE. It was running pretty well until recently. There were some updates for various packages and after installing them Howdy seems to showing various error in howdy test feed also Howdy is taking more time to start than it used to. Please see the bellow error that's showing in terminal when I run `sudo howdy test,` 

[https://paste.ubuntu.com/p/zfJXmGqfFP/](url)

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):
Manjaro KDE
Howdy version:
2.5.1-2

boltgolt, 2020-01-07:
Seems to be a gstreamer issue, could you try switching recorders?

Sobuj53, 2020-01-10:
Hi thank you for your reply, can you please elaborate a little bit, I didn't understand what do you mean by "switching recorders".

boltgolt, 2020-01-10:
In Howdy, a recorder is another program that handles the fetching of the frames from the camera for us. By default this is OpenCV, and you seem to have gstreamer errors from that recorder

You could try to change [this line](https://github.com/boltgolt/howdy/blob/master/src/config.ini#L60) to `ffmpeg` by running `sudo howdy config`

Sobuj53, 2020-01-10:
Thank you very much. This resolved the issue, also the speed of detection has improved noticeably.
For those with the same error, if you are running Arch based distro. then you can follow this, 
1. Install python-ffmpeg, `yay -S python-ffmpeg` (assuming you have yay installed)
2. Change recorder plug-in to `ffmpeg` from `opencv`, You could try to change this [line](https://github.com/boltgolt/howdy/blob/master/src/config.ini#L60) to ffmpeg by running `sudo howdy config`

closed 2020-01-10 by Sobuj53