boltgolt/howdy #302: Howdy locks up system on login

Seltonu, 2020-01-08:
When installing the Plasma desktop on Pop!_OS, upon logging in the entire system freezes. The camera flashes on AFTER needing to manually type in my password, and the system needs a hard reboot to return to the login screen, where the same issue occurs again. I fully believe this to be howdy, because disconnecting the camera allows me to login without any crashing. I've reproduced this on two of my computers.

Linux distribution (if applicable): Pop!_OS 19.10

Howdy version: latest

boltgolt, 2020-01-09:
Hm, could you try [this](https://github.com/boltgolt/howdy/issues/28#issuecomment-396231042)?

ATragicEnding, 2020-01-19:
@boltgolt I've tested the solution posted in the link you gave OP. It does not work from my end. I'm on Kubuntu 19.10.

boltgolt, 2020-01-19:
I think you have a different issue than op does, maybe #208 matches your issue more closely?

ATragicEnding, 2020-01-19:
Both issues seem pretty similar as both locks me out of my session. I can only login by disabling Howdy altogether. I have also tried the solutions posted in #208 but to no avail.

ATragicEnding, 2020-01-19:
I can provide logs, if it could help you.

davi-ro, 2020-03-05:
I'm using Kde Neon and also had troubles logging in with howdy. I started to search for possible solutions and the only two solutions that worked for me were:
sudo chmod -R 755 /lib/security/howdy
The cmd abose solved logging in from lockscreen, but did not solve the logging from the first Login Screen (the one that appears after every restart, after every restart had to cover the camera completely to login by password).
So i found another solution to change no_confirmation to be true in howdy config
sudo howdy config
Type the cmd above and search for no_confirmation string, by default was false so i've set it to true, restarted the os and everything works like a charm now in kde neon with howdy login.

Seltonu, 2020-06-09:
> Hm, could you try [this](https://github.com/boltgolt/howdy/issues/28#issuecomment-396231042)?

Sorry for the delays in testing, I had stopped trying to use Plasma altogether after wiping my system. I've now testing on Pop!_OS 20.04 and can say the issue is similar but different now. Instead of hard freezing the system, I'm just completely unable to get passed the login screen. The suggestion you linked DOES work, with a couple things to note:

-Without the fix, the IR camera does start but it doesn't log me in. Typing my password doesn't work either. Unplugging my webcam and rebooting fixes the issue.
-With the provided fix, chmod alone does not solve the issue, editing the config file is needed too.
-Minor note: With the fix my IR camera still doesn't fire automatically like with Gnome, I need to hit enter on my user profile to start it.

Hopefully I can help with any other debugging if I don't wipe my system in frustration with Plasma soon!

nekromoff, 2020-12-08:
Confirming this issue on KDE Neon 20.04. If camera is available, howdy effectively blocks any login and hard reboot is required.
If camera is disabled (or lid is down), password login works.
SDDM used.

matteosemplice, 2021-02-23:
Hi. Maybe this is related.

I have just installed howdy in Debian and, in order to make it work, I had to
1. change file permissions in /lib/security/howdy
2. swap the order of modules in /etc/pam.d/common-auth to make it look like
# here are the per-package modules (the "Primary" block)
auth    [success=2 default=ignore]      pam_unix.so nullok_secure try_first_pass
auth    [success=1 default=ignore]        pam_python.so /lib/security/howdy/pam.py

Doing only 1 or only 2 did not work (kde with sddm).
In this way, if I just hit enter at the password prompt, the camera turns on and howdy authenticates me.

ForgotMyPass2, 2021-04-10:
> Hi. Maybe this is related.
>
> I have just installed howdy in Debian and, in order to make it work, I had to
>
>     1. change file permissions in /lib/security/howdy
>
>     2. swap the order of modules in [...]

This worked for me. Thanks!

boltgolt, 2021-04-13:
Could try if changing only the file permissions and not swapping those lines is good enough

MoulderUtes, 2021-08-15:
> Hi. Maybe this is related.
>
> I have just installed howdy in Debian and, in order to make it work, I had to
>
>     1. change file permissions in /lib/security/howdy
>
>     2. swap the order of modules in [...]

I did this and it fixed the login but now sudo does not even try howdy it justs asks for the password. Any ideas?

zeiktuvai, 2021-08-20:
Hello Everyone!  I am also having a similar issue with Howdy.  I am recently coming back to Linux from windows (don't like the direction MS is taking win 11) after about 10 years.  I am glad to sea there are many great open source projects like this to gain feature parity.  I am also still learning, so I apologize if I do not understand right away.

So, I have a Dell XPS 17 9710 with Kubuntu 21.04 and I have howdy working fine, and was able to get the IR blaster working as well.  The problem I have is that when I try to log in at the login prompt (SDDM I believe) it accepts my password and then does a howdy authentication.  But what happens is the screen dims, howdy does its work and after the camera turns off nothing happens.  The mouse works, but I can't click anything or log in, so I end up having to Ctrl + Alt + F3 to get into another terminal, log in (which howdy does right away) and restart.

I have tried the auth fix in the post above (though mine has three entries) and the fix mentioned here in #28.  The only way I can log in at the login screen is to cover the camera.  Sudo and all other logins work fine.

Thank you for any help you can give!!!

dovepa, 2021-12-26:
hello i have the same bug :) as zeiktuvai  

> Hello Everyone! I am also having a similar issue with Howdy. I am recently coming back to Linux from windows (don't like the direction MS is taking win 11) after about 10 years. I am glad to sea [...]

einsteinarbert, 2023-09-30:
> Hi. Maybe this is related.
>
> I have just installed howdy in Debian and, in order to make it work, I had to
>
> 1. change file permissions in /lib/security/howdy
> 2. swap the order of modules in [...]

When I did that, I cannot login anymore from shell TTY, Must to go into rescue mode for revert it.
In the dark room, I must to disable camera by my Hand and login by password, so SSDM will work (ubuntu 22.04 KDE)

closed 2026-11-16 by boltgolt