boltgolt/howdy #303: How to access the IR camera using python?

Sailesh009, 2020-01-10:


Sailesh009, 2020-01-10:
Hi BoltGolt

I need a little help. I want to access the IR camera using a python script and earlier i was unable to do that, using OpenCV. But while installing howdy, i observed that it was able to switch on my IR emitters. Could you please help me out on how to access the IR camera using FFMPEG or something else?

boltgolt, 2020-01-10:
It's quite easy, the IR camera is simple another video input. Om my system, that's `/dev/video2`

closed 2020-01-10 by boltgolt

Sailesh009, 2020-01-10:
Hi BoltGolt

I am not able to access on /dev/video2 even though it exists. It says "OPENCVIO error: unsupported format". But while installing Howdy, I did observe that it was able to switch on the emitters and the IR camera. I just want to know how you were able to do that so that I can leverage the same using a python script :)

boltgolt, 2020-01-10:
You can try using the [command used during installation](https://github.com/boltgolt/howdy/blob/master/debian/preinst#L76)

Sailesh009, 2020-01-10:
Thank you Mr.BoltGolt. I was indeed able to switch on the emitter and the camera. But could you now guide a little bit on how to snap a IR picture from it?

boltgolt, 2020-01-10:
The `-o /dev/null 1>/dev/null 2>/dev/null` part discards all output, change it so something like `-o /tmp/ir.jpg` to save the image

Sailesh009, 2020-01-10:
After executing the above, it returned the following error - 

 **neither audio nor video format specified/found**

boltgolt, 2020-01-10:
What exactly are you executing?

Sailesh009, 2020-01-10:
**streamer -t 1:0:0 -c /dev/v4l/by-path/pci-0000:00:14.0-usb-0:4.2:1.2-video-index0 -b 16 -f rgb24 -o outputir.jpg**

where **"pci-0000:00:14.0-usb-0:4.2:1.2-video-index0"** is the IR camera deviceID I found using your program.

boltgolt, 2020-01-10:
Try this: `streamer -t 0:0:1 -c /dev/v4l/by-path/pci-0000:00:14.0-usb-0:4.2:1.2-video-index0 -b 16 -o outfile.jpeg`

Sailesh009, 2020-01-10:
I tried executing the above command. It resulted in a .jpeg file but I am unable to open that file in any application in any application.

I just discovered one thing, this command - 

**streamer -t 1:0:0 -c /dev/v4l/by-path/pci-0000:00:14.0-usb-0:4.2:1.2-video-index0 -b 16 -f rgb24 -o outputir.avi**

is successfully creating a video from the IR camera and I am able to see it properly.