boltgolt/howdy #304: Eyes closed unlock?

danielzgtg, 2020-01-11:
Does Howdy work with your eyes closed? I don't see any configuration options for it.

Windows Hello does work with your eyes closed and that's a bad thing. If Howdy can prevent logins with your eyes closed then it would be more secure than Windows Hello and that should be stated as a feature.

boltgolt, 2020-01-14:
Howdy does not detect closed eyes at this time, but the underling library has [options to detect face features](https://www.pyimagesearch.com/2017/04/03/facial-landmarks-dlib-opencv-python/) and a config option could be build from that. This would be a very significant performance hit though. 

I'd accept any PRs implementing this and if enough people want it i might build it myself as well.

Doomsdayrs, 2020-03-01:
I'd second an option to enable/disable such detentions. Even at the cost of performance

ardevd, 2020-05-26:
I'd want this added as well!

yashahir, 2020-07-03:
Yeah, just make it toggle-able and off by default so it doesn't cause people who don't read change logs any problems. I think it's a great idea and makes it much more secure!

alexanderadam, 2020-07-21:
Eyes closed should probably be the minimum.
It would be better if it would also be attention-aware. See [this description about Face ID](https://support.apple.com/en-us/HT208108#:~:text=Face%20ID%20is%20even%20attention,a%20passcode%20on%20your%20device.) for instance:

> It recognizes if your eyes are open and your attention is directed towards the device. This makes it more difficult for someone to unlock your device without your knowledge (such as when you are sleeping).

fatkaratekid, 2021-10-03:
I'll take a look into this, it is somewhat related to #433