boltgolt/howdy #305: Camera not recognized automatically

frenzymadness, 2020-01-15:
Hello.

My camera is not recognized automatically during installation. I have three devices:
* /dev/media0
* /dev/video0
* /dev/video1

They all point to the same default camera in my ThinkPad X1 Carbon and when testing with `cheese` it seems that they all work the same.

The problem also is that there is no direct documentation although it seems that it happens kinda often. The [Arch wiki page](https://wiki.archlinux.org/index.php/Howdy#Setup_Howdy_to_start_when_needed) helped me a lot. With the instructions there I can test that my camera works (using `cheese`) and configure it manually. It would be nice to add these instructions for manual testing/configuration to the wiki.

Linux distribution (if applicable): Fedora 31

Howdy version: howdy-2.5.1-2.fc31

Have a nice day.

boltgolt, 2020-01-15:
The debian installer is the only one to automatically set the camera if i recall correctly, might be possible to do that in fedora too. Either way, better documentation is needed

dshamany, 2020-03-12:
type `sudo howdy -U [username] config` and change the video stream manually. Mine was /dev/video2 that worked.

closed 2021-04-02 by boltgolt