boltgolt/howdy #306: Install fails with 'repository . . . does not have a Release file" Ubuntu 20.04

chiechis, 2020-01-22:
_Please describe the issue in as much detail as possible, including any errors and traces_

During "add-apt-repository" I get:
. . .
Hit:4 http://security.ubuntu.com/ubuntu focal-security InRelease                                 
Err:5 http://ppa.launchpad.net/boltgolt/howdy/ubuntu focal Release                               
  404  Not Found [IP: 91.189.95.83 80]
Hit:6 http://us.archive.ubuntu.com/ubuntu focal-updates InRelease                  
. . . and then at the end: 
Reading package lists... Done                                                                    
E: The repository 'http://ppa.launchpad.net/boltgolt/howdy/ubuntu focal Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
. . .
Similar errors occur during "apt update", and the package is not found during "apt install howdy".
Referenced apt-secure manpage says  I can force it by setting "Acquire::AllowInsecureRepositories" to Yes, is that the preferred fix?
----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu 20.04

Howdy version: 2.5.1

boltgolt, 2020-01-22:
The issue was that there was no release for 20.04 in the PPA yet, it's building right now so try again in about an hour

closed 2020-01-22 by boltgolt

MkadirDev, 2021-05-03:
If you check the contents of the repository with a web browser, you'll see that this repository does not have a directory for release named bionic.

This PPA seems to have ended support for release bionic or never had one. The repository name includes the part "stable-daily" which suggests it should have had frequent updates, but according to the "last modified" times indicated, this PPA has had no updates since 2017-07-21.

All the versions of Debian starting from release stretch (Debian 9, the current oldstable) now contain VLC 3.0 in the standard repositories, so adding a PPA is no longer necessary if VLC 3.0 is what you want. Since Ubuntu is based on Debian, the current Ubuntu versions will have at least that version too.

If you want nightly builds of the latest versions of VLC, you might want to use the "master-daily" part of the repository instead of no-longer-updated "stable-daily". However, if you look at the status page of the master-daily PPA, it seems the latest VLC has just recently failed to build (1 hour 10 minutes ago as of this writing).

Now you'll have to first decide what you want to do, and update the repository configuration of your system accordingly. Look into /etc/apt/sources.list file and any files in /etc/apt/sources.list.d/ directory to find the URL listed in the original error message.

If you simply wish to stop using this repository, you can comment out the line with this URL (or if it's in its own file in sources.list.d directory, just delete the file).

Since the question tags say debain, the original repository is for Ubuntu, and you don't explicitly say which distribution you're using, I cannot recommend any particular replacement.

boltgolt, 2021-05-03:
@MkadirDev are you commenting on the right repo? Howdy has had a Bionic release in Launchpad since my comment above and the latest version, 2.6.1, has been released for the series on 2020-09-02. I think this comment was meant for a VLC repo?

boltgolt, 2021-05-03:
@MkadirDev This is actually much weirder than that, the comment is copy-pasted from a random [stackexchange answer](https://unix.stackexchange.com/a/578647/207093), with a typo added to it ("debian" -> "debain"). Did you manually retype this and post it here? Why post this on an old issue that's only tangentially related?

stefanpi74, 2025-01-20:
@boltgolt I'm getting pretty much the same error;
The repository 'https://ppa.launchpadcontent.net/boltgolt/howdy/ubuntu oracular Release' does not have a Release file. 
I'm on Ubuntu 24.10.