boltgolt/howdy #307: Install Issue

kevmullet100, 2020-01-24:
_Please describe the issue in as much detail as possible, including any errors and traces_

When installing on fresh install of Elementary OS (deb base) getting errors with dpkg and/or bzip
Installing via ppa, sudo apt install howdy
Have searched the issues and cant find anything similar
Any help would be much appreciated!

Setting up libopencv-dev (3.2.0+dfsg-4ubuntu0.1) ...
Setting up howdy (2.5.1) ...

>>> Upgrading pip to the latest version

Collecting pip
  Downloading https://files.pythonhosted.org/packages/54/0c/d01aa759fdc501a58f431eb594a17495f15b88da142ce14b5845662c13f3/pip-20.0.2-py2.py3-none-any.whl (1.4MB)
    100% |████████████████████████████████| 1.4MB 309kB/s 
Installing collected packages: pip
  Found existing installation: pip 9.0.1
    Not uninstalling pip at /usr/lib/python3/dist-packages, outside environment /usr
Successfully installed pip-20.0.2

>>> Downloading and unpacking data files

Downloading 3 required data files...
Unpacking...
bzip2: Can't open input file *.bz2: No such file or directory.
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for libc-bin (2.27-3ubuntu1) ...
Processing triggers for doc-base (0.10.8) ...
Processing 3 added doc-base files...
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):

Howdy version:

boltgolt, 2020-01-24:
Did you try to run the installation a second time? What does `ls /lib/security/howdy/dlib-data` output?

kevmullet100, 2020-01-24:
Tried to reinstall a load of times earlier in the day, no dice.
Just re-tried and it worked, pulled down some files after the  "Downloading 3 required data files..." message. 
Exactly the same install command, following a purge.
Weird but working fine now. 
Please close the issue

Thanks

boltgolt, 2020-01-25:
Yeah it's weird. Very very rarely wget just does not download anything and the installation fails, i'm guessing it's an environment issue somehow. Good to hear that it's fixed!

closed 2020-01-25 by boltgolt

kevmullet100, 2020-03-29:
I know we closed this a while ago, but just to add more to this.
I've just reimaged my laptop and am experiencing the same issues.
A bit more digging seems to be a DNS problem for raw.githubusercontent resolving to localhost for some reason...
Added a static entry to /etc/hosts for raw.githubusercontent.com resolving to  151.101.84.133 seems to fix it

XXX@XPS:/home$ cat /lib/security/howdy/dlib-data/Readme.md 
Download and unpack `dlib` data files from https://github.com/davisking/dlib-models repository:
```shell
wget https://github.com/davisking/dlib-models/raw/master/dlib_face_recognition_resnet_model_v1.dat.bz2
wget https://github.com/davisking/dlib-models/raw/master/mmod_human_face_detector.dat.bz2
wget https://github.com/davisking/dlib-models/raw/master/shape_predictor_5_face_landmarks.dat.bz2
bunzip *bz2
```
XXX@XPS:/home$ bash /lib/security/howdy/dlib-data/install.sh 
Downloading 3 required data files...
Unpacking...
bzip2: Can't open input file *.bz2: No such file or directory.
XXX@XPS:/home$ ls
XXX
XXX@XPS:/home$ wget https://github.com/davisking/dlib-models/raw/master/dlib_face_recognition_resnet_model_v1.dat.bz2
--2020-03-29 16:09:48--  https://github.com/davisking/dlib-models/raw/master/dlib_face_recognition_resnet_model_v1.dat.bz2
Resolving github.com (github.com)... 140.82.118.3
Connecting to github.com (github.com)|140.82.118.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/davisking/dlib-models/master/dlib_face_recognition_resnet_model_v1.dat.bz2 [following]
--2020-03-29 16:09:48--  https://raw.githubusercontent.com/davisking/dlib-models/master/dlib_face_recognition_resnet_model_v1.dat.bz2
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 0.0.0.0, ::
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|0.0.0.0|:443... failed: Connection refused.
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|::|:443... failed: Connection refused.
XXX@XPS:/home$ ping raw.githubusercontent.com
PING raw.githubusercontent.com (127.0.0.1) 56(84) bytes of data.
64 bytes from localhost (127.0.0.1): icmp_seq=1 ttl=64 time=0.010 ms
64 bytes from localhost (127.0.0.1): icmp_seq=2 ttl=64 time=0.049 ms
^C
--- raw.githubusercontent.com ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1027ms
rtt min/avg/max/mdev = 0.010/0.029/0.049/0.020 ms

boltgolt, 2020-04-01:
Very weird again, thanks for finding a workaround!

Sciroccogti, 2021-07-30:
I ran into this issue too. And a modification of `hosts` may also help.

Please somehow fix it, or add a check after downloading, or something else. 🙂

boltgolt, 2021-07-30:
@Sciroccogti are you beind the great firewall? That seems to be the cause of this

Sciroccogti, 2021-07-30:
> @Sciroccogti are you beind the great firewall? That seems to be the cause of this

Yes, but I think maybe just a little hint will help.

boltgolt, 2021-07-30:
Using a VPN could help, or using the hosts solution.
Unfortunately these files can not be bundled with Howdy due to licensing so pulling them from github is required

devarun, 2023-01-07:
For Ubuntu 20.04, try this https://serverfault.com/a/979014/999626 it worked for me

vchiguru, 2023-09-28:
Stuck at this position for more than one hour and doesn't progress. Tried multiple times but still the same issue.
Plz help.

vivek@chiguru-ROG-M16:~$ sudo apt install howdy
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  nvidia-firmware-535-535.113.01 nvidia-firmware-535-535.54.03 nvidia-firmware-535-535.86.05
Use 'sudo apt autoremove' to remove them.
Suggested packages:
  nvidia-cuda-dev
The following NEW packages will be installed:
  howdy
0 upgraded, 1 newly installed, 0 to remove and 23 not upgraded.
Need to get 0 B/38.9 kB of archives.
After this operation, 166 kB of additional disk space will be used.
Selecting previously unselected package howdy.
(Reading database ... 307625 files and directories currently installed.)
Preparing to unpack .../archives/howdy_2.6.1_all.deb ...
Starting IR camera check...

Trying "USB2.0 HD UVC WebCam: USB2.0 HD"
One of your cameras should now be on.
Did your IR emitters turn on? [y/N]: y

Starting certainty auto config...

After detection, Howdy knows how certain it is that the match is correct.
How certain Howdy needs to be before authenticating you can be customized.

F: Fast.
Allows more fuzzy matches, but speeds up the scanning process greatly.

B: Balanced.
Still relatively quick detection, but might not log you in when further away.

S: Secure.
The safest option, but will take much longer to authenticate you.

You can always change this setting in the config.
What profile would you like to use? [f/b/s]: b

Unpacking howdy (2.6.1) ...
Setting up howdy (2.6.1) ...

>>> Upgrading pip to the latest version

Requirement already satisfied: pip in /usr/lib/python3/dist-packages (23.0.1)
Collecting pip
  Downloading pip-23.2.1-py3-none-any.whl (2.1 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 2.1/2.1 MB 12.8 MB/s eta 0:00:00
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 23.0.1
    Not uninstalling pip at /usr/lib/python3/dist-packages, outside environment /usr
    Can't uninstall 'pip'. No files were found to uninstall.
  WARNING: The scripts pip, pip3 and pip3.11 are installed in '/usr/local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed pip-23.2.1
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is 
recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv

>>> Upgrading numpy to the latest version

Requirement already satisfied: numpy in /usr/lib/python3/dist-packages (1.24.2)
Collecting numpy
  Obtaining dependency information for numpy from https://files.pythonhosted.org/packages/c4/36/161e2f8110f8c49e59f6107bd6da4257d30aff9f06
373d0471811f73dcc5/numpy-1.26.0-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata
  Downloading numpy-1.26.0-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (58 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 58.5/58.5 kB 2.6 MB/s eta 0:00:00
Downloading numpy-1.26.0-cp311-cp311-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (18.2 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 18.2/18.2 MB 23.3 MB/s eta 0:00:00
Installing collected packages: numpy
  Attempting uninstall: numpy
    Found existing installation: numpy 1.24.2
    Uninstalling numpy-1.24.2:
      Successfully uninstalled numpy-1.24.2
  WARNING: The script f2py is installed in '/usr/local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed numpy-1.26.0
WARNING: Running pip as the 'root' user can result in broken permissions and conflicting behaviour with the system package manager. It is 
recommended to use a virtual environment instead: https://pip.pypa.io/warnings/venv

>>> Downloading and unpacking data files

Downloading 3 required data files...
    
Progress: [ 60%] [#####################################################################...............................................]