boltgolt/howdy #312: Lenovo X1 - Carbon gen 7 - Ir Emitter issues

dockera, 2020-02-09:
Hi There,

Great work on howdy. Had an issue where the Chicony IR emitter was not functioning at all. After some research, compiled the IR toggle app in this repository. This ran effectively under sudo and I was able to use howdy during a session. However, the IR emitter was not toggled back on by default under Ubuntu 19.10. To get around this, I created a service inside systemd (/lib/systemd/system) called chicony.service. To create this, used the following steps

1. move into the appropriate directory - cd /lib/systemd/system
2. create the service file - sudo touch chicony.service
3. edit the service file - sudo nano ./chicony.service
4. edited the service file by inserting the content below

[Unit]
Description=chicony service

[Service]
ExecStart=/path to compiled ir-toggle-command/chicony-ir-toggle-master/chicony-ir-toggle on

[Install]
WantedBy=multi-user.target

6. Set permissions for the service to match the other services in the directory
7. ran -  sudo systemctl start chicony - this starts the service and runs the command
8. ran - sudo systemctl status chicony  - checks that the service ran successfully
9. ran - sudo systemctl enable chicony - sets the service to run at boot

This ensures that the IR emitter is toggled on, allowing howdy to run from boot

Hope someone finds this useful. Found that the emitter only ever flashes once, not sure how to replicate the pattern of usage under Windows whereby windows hello causes it to flash rapidly and repeatedly - this likley affects accuracy etc. 

Would be keen to see the IR toggle function set within howdy, would offer to help but not a real coder :-)

shivindera, 2020-02-20:
Just to add to the above answer. Information I feel could be useful:
- The repo for chicony-ir-toggle is - [chicony-ir-toggle](https://github.com/PetePriority/chicony-ir-toggle)
- If you follow the install steps, the executable file is located in `/usr/local/bin/chicony-ir-toggle`
- When you run `chicony-ir-toggle on`, the ir sensor feed goes through video2, not video0,
- Modify the file `/lib/security/howdy/config.ini` and change the line to `device_path = /dev/video2`
- Run `sudo howdy test` and you will see that the feed is coming from the ir sensor.

Note: I had to modify the dark_threshold to 100 in the config file for it to recognise my face using the IR sensor. @dockera did you have to do something similar?

LEXUGE, 2020-03-11:
I have packaged a NixOS [package](https://github.com/LEXUGE/nixos/blob/master/packages/ir_toggle.nix) and a [module](https://github.com/LEXUGE/nixos/blob/master/modules/ir_toggle.nix) for ir-toggler.
It would work on boot and after resume. Feel free to grab if you need them. Tested/worked on X1 Carbon 2019 20R1 model

rtar, 2020-04-20:
> Hope someone finds this useful. Found that the emitter only ever flashes once, not sure how to replicate the pattern of usage under Windows whereby windows hello causes it to flash rapidly and repeatedly - this likley affects accuracy etc.

Changing resolution to something like 360x360 fixes the issue on ThinkPad P43s. See more for the details: https://github.com/boltgolt/howdy/wiki/Common-issues#camera-ir-emitters-not-turning-on

orclex, 2020-05-12:
Hi, I also have a Lenovo X1C7 and howdy is working out of the box (starting with Ubuntu 19.04 and up to current 20.04). What exactly is chicony-ir-toggle doing and why is it necessary or recommended? Thanks!

rtar, 2020-05-12:
@orclex On some laptops Howdy does not turn on IR emitters. I.e. if you see a red light appearing when Howdy runs near your camera, it means IR emitters are working out of the box and you do not need this application.

If you do not see the red light, Howdy might still work if you are working in a good lightning (i.e. at daytime), but will not work properly in a dark room. This app fixes that issue and gets IR emitters working again.

orclex, 2020-05-13:
Thanks for your explanation. I've never seen a red light before and indeed I got bad results in dark light. I'll give this application a try. It would be great if it could be integrated into howdy...

BTW: what is your experience in changing howdy config for best face detection?

rtar, 2020-05-13:
@orclex You can check your configuration setup easily by running `sudo howdy test`. It will show you what camera shows, and a circle around your face if it detects it.

Changing configuration will help when there is enough light, but will not do anything in dark. You will understand why immediately when you see results of `sudo howdy test` in a dark room (hint: camera does not see anything without IR emitters turned on). With IR emitters working, Howdy will work perfectly with default settings (well, you might need to change camera resolution or IR emitters will blink instead of kept being on until recognition works out).

I think this app is simple enough, so if someone gets attention of Howdy authors (or just do the pull request) they may include it, indeed. I did not care enough, so I just installed it, and included it into `systemctl` so it gets turned on each time the system starts.

Note, that `chicony-ir-toggle` does not turn IR emitters on. Howdy does it. This app just enables them so Howdy can do it.

orclex, 2020-05-13:
@rtar thanks for explanation. It's working now with chicony-ir-toggle on and systemd service file as well as default howdy configuration.

For my further understanding:
I can get possible video resolutions for the camera with
```
# v4l2-ctl -d /dev/video0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'MJPG' (Motion-JPEG, compressed)
                Size: Discrete 1280x720
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x180
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 424x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 848x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 960x540
                        Interval: Discrete 0.033s (30.000 fps)
        [1]: 'YUYV' (YUYV 4:2:2)
                Size: Discrete 1280x720
                        Interval: Discrete 0.100s (10.000 fps)
                Size: Discrete 320x180
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 320x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 352x288
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 424x240
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x360
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 640x480
                        Interval: Discrete 0.033s (30.000 fps)
                Size: Discrete 848x480
                        Interval: Discrete 0.050s (20.000 fps)
                Size: Discrete 960x540
                        Interval: Discrete 0.067s (15.000 fps)

```

as well as

```
# v4l2-ctl -d /dev/video2 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'GREY' (8-bit Greyscale)
                Size: Discrete 640x360
                        Interval: Discrete 0.067s (15.000 fps)
                        Interval: Discrete 0.033s (30.000 fps)
```

The default howdy config is 
`max_height = 320`

To be more precise I'd like to try higher resolutions like 360. I'm not sure if I'm limited to video2 for ir or video0 resolutions for the camera. Setting any other height seems to be ignored in _howdy test_. So I'm not sure if I can optimize the resolution...

rtar, 2020-05-13:
Did you try changing `frame_width` and `frame_height` besides `max_height`?

orclex, 2020-05-14:
Yes I tried the other options, but it did not change anything. Furthermore frame_heigth/width is set to -1 as default which should use the largest possible profile. But if these options are only for camera and not IR then I do not want to change them. ;-)

orclex, 2020-05-20:
I've noticed that after resume from suspend the first try to unlock session does not use the ir emitter. On another try to unlock session with howdy the ir emitter turns on and works.
So I guess the chicony.service starts too late from suspend. Is there a way to prioritize the service higher and start earlier after wake from suspend?

reuseman, 2020-12-14:
The same applies for the Thinkpad T14s

EmixamPP, 2020-12-18:
I think my repo :  https://github.com/EmixamPP/linux-enable-ir-emitter (it's more or less a fork of chicony-ir-toggle) can help you to solve your problems, A detailed tutorial in the readme helps you to adapt the script to your camera.