boltgolt/howdy #315: The ADD isnt working

EL-dog, 2020-02-23:
_Please describe the issue in as much detail as possible, including any errors and traces_

 sudo howdy add
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/lib/security/howdy/cli/add.py", line 49, in <module>
    face_encoder = dlib.face_recognition_model_v1(path + "/../dlib-data/dlib_face_recognition_resnet_model_v1.dat")
RuntimeError: Unable to open /lib/security/howdy/cli/../dlib-data/dlib_face_recognition_resnet_model_v1.dat for reading.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu

Howdy version:

boltgolt, 2020-02-24:
Try rerunning the install script in `/lib/security/howdy/dlib-data`

Erudyx, 2020-02-25:
same problem here on Ubuntu 20.04, with a different error.
i can enter a token, and then, i have this output:

`Adding face model for the user erudyx
Enter a label for this new model [Initial model] (max 24 characters): Erudyx#0

Please look straight into the camera
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 89, in <module>
    import cli.add
  File "/lib/security/howdy/cli/add.py", line 147, in <module>
    gsframe = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.2.0) /io/opencv/modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor' `

running `sudo howdy add` in `/lib/security/howdy/dlib-data` doesnt seems to change anything

boltgolt, 2020-02-25:
What I meant by "running the install script" was executing the bash script in that folder, I believe it's simply called install, so running `./install.sh` in that folder should do it

MurzNN, 2020-02-28:
Got same problem on Ubuntu 19.10, re-running `./install.sh` not help:

```
root@ubuntu:/lib/security/howdy/dlib-data# ./install.sh 
Downloading 3 required data files...
dlib_face_recognition_resnet_model_v1.dat.bz2                       100%[==================================================================================================================================================================>]  20.44M   548KB/s    in 36s     
mmod_human_face_detector.dat.bz2                                    100%[==================================================================================================================================================================>] 678.41K   839KB/s    in 0.8s    
shape_predictor_5_face_landmarks.dat.bz2                            100%[==================================================================================================================================================================>]   5.44M   634KB/s    in 8.0s    
Unpacking...
bzip2: Output file dlib_face_recognition_resnet_model_v1.dat already exists.
bzip2: Output file mmod_human_face_detector.dat already exists.
bzip2: Output file shape_predictor_5_face_landmarks.dat already exists.
```

What else we can try to debug or fix this problem?

MurzNN, 2020-02-28:
Seems the problem is with missed config file, here is output of `test` command:
```
$ sudo howdy test

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

Unknown camera, please check your 'device_path' config value.

Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 101, in <module>
    import cli.test
  File "/lib/security/howdy/cli/test.py", line 117, in <module>
    frame = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.2.0) /io/opencv/modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'
```

boltgolt, 2020-02-28:
Running `./install.sh` was meant for the OP, both of your issues are not related to his. Your errors point to an unreadable camera stream. Changing it to index 1 probably switched it from the IR camera to the normal visible light one?

YanhuiJessica, 2021-03-10:
> Try rerunning the install script in `/lib/security/howdy/dlib-data`

Ubuntu 20.04 fixed RE after running `install.sh` :+1:

closed 2021-04-02 by boltgolt