boltgolt/howdy #317: cannot install

ryanhardt41, 2020-02-27:
i've tried installing on 4 different machines over 2 distros, all with IR sensors and a working camera, all got the same error listed in the image (sorry if its a bit small):

[image]

problem persisted over a month, please help !!

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Zorin OS 15 (based off Ubuntu 18.04 LTS), Kubuntu 18.04 LTS

Howdy version: none

ryanhardt41, 2020-02-27:
btw everything goes smoothly until when i actually try to install howdy itself

boltgolt, 2020-02-27:
Have you tried googling the error? This is a common issue with dpkg that can appear after power loss or similar interruptions

ryanhardt41, 2020-02-28:
alright, i solved the problem by running sudo dpkg --configure -a and then reinstalling. thanks for your help !!

boltgolt, 2020-03-01:
Did you try rebooting?

closed 2021-04-02 by boltgolt